TheThingsProducts/gateway #4: build automation

grahamehorner, 2018-02-05:
please publish details of where to download the artefacts from a build automation for those that can'/don't wish to configure a CI environment but wish to deploy images to the gateway and help improve/debug the firmware/

htdvisser, 2018-02-14:
[Travis CI](https://travis-ci.org/TheThingsProducts/gateway) now automatically builds the branches in this repository. I added the location of the build artifacts to the [README](https://github.com/TheThingsProducts/gateway/blob/master/firmware/README.md#firmware-releases).

If you enable Travis CI on your fork of this repository and set your own `AZURE_STORAGE_ACCOUNT` and `AZURE_STORAGE_KEY` variables in Travis CI, build artifacts will automatically be uploaded to your own Azure Storage Account.

Anyone who wants to build the firmware locally can follow the [Local Build](https://github.com/TheThingsProducts/gateway/blob/master/firmware/README.md#local-build) and [Docker Build](https://github.com/TheThingsProducts/gateway/blob/master/firmware/README.md#build-in-docker) instructions which I added 2 days ago.

closed 2018-02-14 by htdvisser

grahamehorner, 2018-02-14:
@htdvisser great work