TheThingsProducts/gateway #5: Gateway Boot Issues

grahamehorner, 2018-02-07:
My gateway was effected by the reboot issue; now the gateway fails to boot, the power LED is solid; while the second LED is flashing rapidly ?

grahamehorner, 2018-02-07:
I have attempt to update via SD card without success, there is no output via serial port logs

johanstokking, 2018-02-07:
Duplicate of #1

grahamehorner, 2018-02-07:
This is not a duplicate of #1 as I cant connect to the gateway at all the AP is not visible and the gateway is not rebooting but is stuck with regards the boot sequence with power LED solid and 2nd LED flashing rapidly; which is quite a different set of issues to the Reboot Loop

grahamehorner, 2018-02-07:
Please reopen

grahamehorner, 2018-02-07:
I'm unsure this is a soft/firmware issue, as after many attempts and reboots the gateway is now showing up as an AP and I'm able to connect; and the gateway now looks to have successfully updated it's firmware

Version Info |  -- | -- Hardware: | v1
Bootloader: | r1-7167873a (2017-06-02T13:48:18Z)
Firmware: | v1.0.1-0407f7eb (2018-02-06T10:58:41Z)

however the #1 is still present; If hardware issues are still to be logged; this IMHO is definitely an odd one? may be a dry joint or something? physical with the gateway

grahamehorner, 2018-02-16:
The gateway is now back to not showing up as an AP or updating, this time  tried 14 reboots, nothing still not seeing the gateway as an AP so can’t verify if the status?  LEDs are same as above

grahamehorner, 2018-02-20:
After leaving the gateway off for approx. 3days my gateways is still in an unstable state where I'm unable to see the gateways as an access point or update the firmware via the SD card after a full reset ?

grahamehorner, 2018-02-24:
other update booted the gateway this morning, and this time it shows up as an AP; will try and update firmware

grahamehorner, 2018-02-25:
Update seemed to work, then gateway was stuck in reboot loop 😭

grahamehorner, 2018-02-28:
my gateway is shown that the card is ND this never changes even after rebooting or full reset, when the gateway decides to start? Please can any one advise?!

KrishnaIyer, 2018-09-27:
@grahamehorner: Please let us know if you still face this issue after  upgrading to firmware `v1.0.5` or if we can close this.

grahamehorner, 2018-09-27:
@KrishnaIyer I'm unable to perform any update to the gateway as detailed in numerous posts on forums, the gateway is unresponsive to any attempt to update or connection via FDTI, its now an as much use as an expensive door stop or chocolate fireguard :'(  very very frustrating especially when it has been so long without a resolution or an offer of exchange/replacement unit