TheThingsProducts/gateway #8: Gateway loses network connection and needs power cycle to recover

avbentem, 2018-02-12:
My gateway lost its WiFi connection over an hour ago, and could not recover without cycling its power, like others have reported on the forum as well. Some observations:

## While failing to reconnect

- I think it had been running for about 20 hours. (Certainly not a full day; I took a picture at 9 PM last night, when it was not connected; at around 6 PM today it showed as "Last seen: 54 minutes ago".)

- It might have been in AP mode (LED 2 flashing slow and fast).

- Plugging in a debug cable restarted the gateway.

- The unexpected reset of the gateway did not resolve this.

- After the reset, it kept reporting things like:

    ```text
    INET: Trying to connect to WiFi router again (after 120 seconds)
    INET: State change to 0
    MON: SYS Stack size: 3959
    MON: heap usage: 148KB (148KB), free: 191KB
    INET: No Ethernet and WiFi link (after 10 seconds)
    INET: State change to 1
    ```

- The FRITZ!Box 7360 WiFi AP did not log any connection attempts.

- Restarting the WiFi AP did not resolve this. All other devices (Sonos, Apple TV, Chromecast, laptop, tablet, telephone) reconnected just fine.

- Attaching an ethernet cable did not resolve this. The connector's LEDs were flashing, but I forgot to check if the router showed a connection.

- Finally cycling power resolved it.

## During normal operation

- When connected to ethernet and unplugging the cable, it nicely recovers by connecting to WiFi. (LED 2 flashing for some time, while LED 1 and 3 are solid.)

- Restarting the AP when the gateway is connected to its WiFi, makes the gateway reboot:

    ```text
    LORA: Accepted packet
    MQTT: Sending UPLINK failed: -1

    MAIN: MQTT error

    MAIN: Leaving state 5
    SNTP: State change from 0 to 0
    SNTP: State change from 0 to 0
    ```

    Next, when it cannot connect to WiFi after restarting, it starts in AP mode (LED 2 flashing) but eventually connects to WiFi again, and all is good.

avbentem, 2018-02-14:
It kept running for 36 hours, but eventually got into a non-recoverable network problem again. The new logs shows:

- It _can_ recover from some MQTT problems (10:07 this morning, Amsterdam time).

- Sometimes it cannot recover from MQTT problems, even though it has an internet connection and can nicely fetch its configuration from the internet.

So I wonder:

- Could the non-recoverable problem have started by MQTT problems on the server, around 2018-02-14 17:05:15 (Amsterdam time), and be different MQTT problems than the one it could recover from?

## Nicely recovering from `MQTT: Sending status failed` (after running for 36 hours)

- Running my own firmware, with debug logging enabled. (See PR #7.)

- Logging started 2018-02-12 22:42:10. I'm not sure how long the gateway was running then, but not for very long.

- It failed about 36 hours later, `MQTT: Sending status failed`, but recovered. All logs here have `MON:`-lines removed for brevity:

	```text
	[2018-02-14 10:06:11] MQTT: Sending UPLINK OK
	[2018-02-14 10:06:22] MQTT: Sending status packet
	[2018-02-14 10:06:22] MQTT: Sending status succeeded: 1351
	[2018-02-14 10:06:47] LORA: Packet dropped! Bad CRC
	[2018-02-14 10:06:52] MQTT: Sending status packet
	[2018-02-14 10:06:52] MQTT: Sending status succeeded: 1352
	[2018-02-14 10:07:07] LORA: Packet dropped! Bad CRC
	[2018-02-14 10:07:22] MQTT: Sending status packet
	[2018-02-14 10:07:22] MQTT: Sending status succeeded: 1353
	[2018-02-14 10:07:52] CB: MQTT: Sending status packet
	[2018-02-14 10:07:54] MQTT: Sending status failed
	[2018-02-14 10:07:54] 
	[2018-02-14 10:07:54] MAIN: MQTT error
	[2018-02-14 10:07:55] 
	[2018-02-14 10:07:55] MAIN: Leaving state 5
	[2018-02-14 10:07:57] MAIN: Entering state 6
	[2018-02-14 10:07:57] INET: State change to 0
	[2018-02-14 10:07:57] WIFI: Disabling modules
	[2018-02-14 10:07:57] SNTP: State change from 7 to 8
	[2018-02-14 10:07:57] CB: Disconnect
	[2018-02-14 10:07:57] Head magic match void: trying to free an already freed block, ignore
	[2018-02-14 10:07:57] SNTP: State change from 8 to 1
	[2018-02-14 10:07:57] WIFI: Entering state 3
	[2018-02-14 10:07:58] WIFI: Enabling modules for client
	[2018-02-14 10:07:58] WIFI: Entering state 6
	[2018-02-14 10:07:58] WIFI: IP Address: 0.0.0.0 
	[2018-02-14 10:08:00] CB: INET: Gateway has WiFi
	[2018-02-14 10:08:00] INET: State change to 2
	[2018-02-14 10:08:00] INET: Connected to a network, waiting for DHCP lease, checking validity with ping
	[2018-02-14 10:08:01] SNTP: State change from 1 to 2
	[2018-02-14 10:08:04] CB: MON: SYS Stack size: 2833
	[2018-02-14 10:08:05] INET: State change to 3
	[2018-02-14 10:08:05] INET: Ping probe
	[2018-02-14 10:08:05] INET: Error sending probe on Eth
	[2018-02-14 10:08:05] LORA: Accepted packet
	[2018-02-14 10:08:06] SNTP: State change from 2 to 6
	[2018-02-14 10:08:09] INET: Ping Timeout
	[2018-02-14 10:08:09] INET: Ping probe
	[2018-02-14 10:08:09] INET: Error sending probe on Eth
	[2018-02-14 10:08:11] LORA: Accepted packet
	[2018-02-14 10:08:13] INET: Ping Timeout
	[2018-02-14 10:08:13] INET: Ping probe
	[2018-02-14 10:08:13] INET: Error sending probe on Eth
	[2018-02-14 10:08:16] CB: Disconnect
	[2018-02-14 10:08:17] INET: Ping Timeout
	[2018-02-14 10:08:17] INET: State change to 1
	[2018-02-14 10:08:17] WIFI: Disabling modules
	[2018-02-14 10:08:17] Head magic match void: trying to free an already freed block, ignore
	[2018-02-14 10:08:17] SNTP: State change from 6 to 1
	[2018-02-14 10:08:17] WIFI: Entering state 3
	[2018-02-14 10:08:18] WIFI: Entering state 4
[... 378 lines cut ...]

	```text
	2018-02-14 18:15:02] **************************
	[2018-02-14 18:15:02] *   The Things Network   *
	[2018-02-14 18:15:02] *      G A T E W A Y     *
	[2018-02-14 18:15:02] **************************
	[2018-02-14 18:15:02] Firmware name: Arjan's v2 (flush UART, extended logging), type: 0, version: 0.0.0, commit: 00000000, timestamp: 0
	[2018-02-14 18:15:02] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
	[2018-02-14 18:15:02] Build time: Feb 11 2018 12:03:25
	[2018-02-14 18:15:02] Reboot reason: 0x40
	```

- ...and all is fine again.

## Full log files

Even more logging in the attached files...

- Gateway log (running my own firmware): [arjan-gateway-20180214.log.zip](https://github.com/TheThingsProducts/gateway/files/1725298/arjan-gateway-20180214.log.zip)

- Router log (incomplete, but showing the troublesome dates): [arjan-fritzbox-20180214.log](https://github.com/TheThingsProducts/gateway/files/1725300/arjan-fritzbox-20180214.log)

## LEDs while failing to connect (just before the manual reset)

[image]

avbentem, 2018-02-18:
Update: it has been running for over 3 days (since 2018-02-14 23:18) now. It has seen a bunch of `MQTT: Sending UPLINK failed` and `MQTT: Sending status failed`, has rebooted 8 times (3 of which for `MAIN: Rebooting gateway for firmware update check`), but has always recovered during the last 3 days.

(Still running my own patched firmware, [being `develop` from February 14th with debug logging enabled](https://github.com/TheThingsProducts/gateway/issues/1#issuecomment-365746884).)

grahamehorner, 2018-02-18:
@avbentem that’s great news, so given the gateway reboots every 24hrs in the code, you’ve got 5 unattend reboots which is a MASSIVE improvement by far, great work

avbentem, 2018-02-18:
To clarify, @grahamehorner, the 5 unwanted reboots occurred after running into a `MAIN: MQTT error` (which, however, does not _always_ imply a reboot; often it just recovers without such reboot and I've not yet discovered in which cases it deems a reboot required), and seem unrelated to reboot issues during activation, like those reported in #1.

I'm eager to test what happens when I make my router disconnect the WiFi, or shutdown the router altogether. But I don't want to interfere with the current test which meanwhile has given me a log file of almost 4 days now, and which may be useful to compare to server side logs, to find the cause of MQTT issues.

grahamehorner, 2018-02-18:
@avbentem that sound like a great plan, wondering if the gateway should have some SNMP added late to allow remote diagnostics and or management;  but it great to hear that progress is begin made

johanstokking, 2018-02-20:
@avbentem do you have the dates and times of this disconnect? We have had some issues with EU endpoints, including MQTT, very recently. This may be server initiated disconnects which the gateway recovered from naturally.

cc @htdvisser

avbentem, 2018-02-20:
Still running, almost 6 days now. And it nicely recovered at about 17:40 GMT+1, without me interfering, from the major EU region problems that [were happening today](https://www.thethingsnetwork.org/forum/t/eu-region-routing-services-major-outage/12768). 

During the problems I got into regular reboots with `RQMQTT: Connection failed` followed by `MAIN: MQTT error` and `Reboot reason: 0x10`. 

At other times in the recent logs I see `MQTT: Connection lost` (rather than `failed`) like the following:

```text
LORA: Accepted packet
MQTT: Sending UPLINK OK
LORA: Accepted packet
MQTT: Sending UPLINK OK
MQTT: Sending status packet
MQTT: Sending status succeeded: 1
MQTT: Connection lost

MAIN: MQTT error
```

...followed by many:

```text
MQTT: GOT IP: 52.169.76.203
Connecting to: 52.169.76.203
MQTT: Opening socket timed out, restarting
MQTT: GOT IP: 52.169.76.203
Connecting to: 52.169.76.203
MQTT: Opening socket timed out, restarting
MQTT: GOT IP: 52.169.76.203
Connecting to: 52.169.76.203
MQTT: Opening socket timed out, restarting
```

So, it seems `Connection failed` (often) yields a reboot, but `Connection lost` makes the firmware try forever.

avbentem, 2018-02-20:
@johanstokking, here's a full log since February 14th:

[arjan-gateway-20180220.log.zip](https://github.com/TheThingsProducts/gateway/files/1741101/arjan-gateway-20180220.log.zip)

Like earlier, this is using my own firmware, based on the February 14th `develop` branch, only changed to enable debug logging and the debug cable's baud rate. It's been compiled with a free version (or a trial?) of the Microchip XC32 v1.42 compiler.

avbentem, 2018-02-20:
@johanstokking and @htdvisser I've edited my comment above a few times.

avbentem, 2018-03-02:
Quick update: my gateway nicely recovered from today's https://status.thethings.network/incident/45 So, it has not needed any reboot since February 14th.

```text
[2018-03-02 10:28:05] MQTT: Sending UPLINK OK
```

However, half an hour later, it ran into:

```text
[2018-03-02 11:02:43] MQTT: Sending UPLINK OK
[2018-03-02 11:02:45] MON: SYS Stack size: 2847
[2018-03-02 11:02:45] MON: heap usage: 262KB (263KB), free: 77KB
[2018-03-02 11:02:45] MQTT: Connection lost
[2018-03-02 11:02:45] 
[2018-03-02 11:02:45] MAIN: MQTT error
...
[2018-03-02 11:02:58] MQTT: GOT IP: 52.169.76.203
[2018-03-02 11:02:58] Connecting to: 52.169.76.203
[2018-03-02 11:02:59] MQTT: Opening socket timed out, restarting
[2018-03-02 11:02:59] MQTT: GOT IP: 52.169.76.203
[2018-03-02 11:02:59] Connecting to: 52.169.76.203
[2018-03-02 11:03:00] MQTT: Opening socket timed out, restarting
```

Again, it recovered. The WiFi logs show:

```text
[02.03.18 11:10:31] Wireless device deregistered itself (2,4 GHz), THINGS-GATEWAY, IP 192.168.178.53, MAC 00:1E:C0:3A:F6:1B.
[02.03.18 11:10:33] Wireless device registered (2,4 GHz), 72 Mbit/s, THINGS-GATEWAY, IP 192.168.178.53, MAC 00:1E:C0:3A:F6:1B.
```

...and the gateway runs fine again:

```text
[2018-03-02 11:10:38] MQTT: TLS ready: Connect MQTT
[2018-03-02 11:10:38] LORA: Accepted packet
[2018-03-02 11:10:38] RQMQTT: Connected
[2018-03-02 11:10:38] 
[2018-03-02 11:10:38] *************************
[2018-03-02 11:10:38] MAIN: Gateway bridging
[2018-03-02 11:10:38] *************************
[2018-03-02 11:10:38] 
[2018-03-02 11:10:38] MQTT: Sending UPLINK OK
[2018-03-02 11:10:38] MQTT: Sending UPLINK OK
```

...until, unfortunately:

```text
[2018-03-02 11:39:28] LORA: Accepted packet
[2018-03-02 11:39:31] MQTT: Sending UPLINK failed: -1
[2018-03-02 11:39:31] 
[2018-03-02 11:39:31] MAIN: MQTT error
```

It then rebooted, and started showing:

```text
[2018-03-02 11:39:50] INET: No Ethernet and WiFi link (after 10 seconds)
...
[2018-03-02 11:41:50] INET: Trying to connect to WiFi router again (after 120 seconds)
[2018-03-02 11:41:50] INET: State change to 0
[2018-03-02 11:42:00] MON: SYS Stack size: 3959
[2018-03-02 11:42:00] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-02 11:42:00] INET: No Ethernet and WiFi link (after 10 seconds)
...
... repeated many, many times
...
[2018-03-02 13:58:02] INET: Trying to connect to WiFi router again (after 120 seconds)
[2018-03-02 13:58:02] INET: State change to 0
[2018-03-02 13:58:03] MON: SYS Stack size: 3959
[2018-03-02 13:58:03] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-02 13:58:12] INET: No Ethernet and WiFi link (after 10 seconds)
```

It has been like this for over two hours now. The WiFi AP is running just fine (same timezone), but for the gateway only shows a single connection attempt:

```text
[02.03.18 11:44:15] Wireless LAN registration failed(2,4 GHz): Access check failed. MAC address: 00:1E:C0:3A:F6:1B.
```

Will post full logs later on, but just in case something special happened around 2018-03-02 11:39:31 Amsterdam time then you might want to remember that?

avbentem, 2018-03-07:
I doubt it helps to add yet another occurrence, but last night it restarted for a firmware check and did not get a network connection until I pressed the mode button a few minutes ago.

```text
[2018-03-06T22:10:30.432Z] LORA: Accepted packet
[2018-03-06T22:10:30.466Z] MQTT: Sending UPLINK OK
[2018-03-06T22:10:31.229Z]
[2018-03-06T22:10:31.231Z] MAIN: Rebooting gateway for firmware update check
[2018-03-06T22:10:35.563Z] MON: SYS Stack size: 2837
[2018-03-06T22:10:35.564Z] MON: heap usage: 262KB (263KB), free: 77KB
[2018-03-06T22:10:40.112Z] SNTP: State change from 0 to 0
[2018-03-06T22:10:40.114Z] SNTP: State change from 0 to 0
[2018-03-06T22:10:40.116Z]
[2018-03-06T22:10:40.116Z]
[2018-03-06T22:10:40.117Z]
[2018-03-06T22:10:40.117Z] **************************
[2018-03-06T22:10:40.118Z] *   The Things Network   *
[2018-03-06T22:10:40.120Z] *      G A T E W A Y     *
[2018-03-06T22:10:40.121Z] **************************
[2018-03-06T22:10:40.122Z] Firmware name: Arjan's v4 (develop, debug, baudrate), type: 0, version: 0.0.0, commit: 00000000, timestamp: 0
[2018-03-06T22:10:40.128Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-03-06T22:10:40.129Z] Build time: Feb 14 2018 23:12:28
[2018-03-06T22:10:40.130Z] Reboot reason: 0x10
[2018-03-06T22:10:40.136Z] BOOT: (persisted info) 6F 72 72 65 01 10 8B 2C E7 16 0D 49 B2 3A 5C B4
[2018-03-06T22:10:40.136Z]
[2018-03-06T22:10:40.137Z]
[2018-03-06T22:10:40.137Z]
[2018-03-06T22:10:40.138Z]
[2018-03-06T22:10:40.138Z] WIFI: Entering state 0
[2018-03-06T22:10:40.139Z] WIFI: Entering SCAN state 0
[2018-03-06T22:10:40.140Z]
[2018-03-06T22:10:40.140Z] MAIN: Initialisation complete
[2018-03-06T22:10:40.141Z] LORA: Changing state from 0 to 0
[2018-03-06T22:10:40.358Z]
[2018-03-06T22:10:40.359Z] MAIN: Leaving state 0
[2018-03-06T22:10:40.360Z] MAIN: Entering state 1
[2018-03-06T22:10:40.362Z] FLASH: Global Block Protection Unlock
[2018-03-06T22:10:40.363Z] FLASH: Manufacturer ID:bf Device Type:26 Device ID:43
[2018-03-06T22:10:40.364Z] FLASH: Status Register[7:0]: 00
[2018-03-06T22:10:40.369Z] FLASH: Configuration Register[7:0]: 08
[2018-03-06T22:10:40.374Z] FLASH: Block-Protection Register[143:0]: 0000 00000000 00000000 00000000 00000000
[2018-03-06T22:10:40.375Z] FLASH: Magic bytes found: wifi config present
[2018-03-06T22:10:40.376Z] FLASH: Magic bytes found: activation data present
[2018-03-06T22:10:40.378Z] FLASH: Magic bytes found: FOTA data present
[2018-03-06T22:10:40.378Z] FLASH: Loading Firmware Data
[2018-03-06T22:10:40.380Z] FLASH: Reading 32 bytes from address 0
[2018-03-06T22:10:40.381Z] FLASH: Loading data succeeded
[2018-03-06T22:10:40.383Z] CNFG: (Firmware HASH (sha256)) BC 56 70 F8 2C 4A 70 13 C4 23 6D 7B 53 BE 61 2D 53 3D 0D 7A F5 78 67 23 85 3C FE 15 22 3D C9 6F
[2018-03-06T22:10:40.384Z] FLASH: Loading WiFi Data
[2018-03-06T22:10:40.385Z] FLASH: Reading 104 bytes from address 4096
[2018-03-06T22:10:40.386Z] FLASH: Loading data succeeded
[2018-03-06T22:10:40.387Z] CNFG: WiFi SSID:      <redacted>
[2018-03-06T22:10:40.388Z] CNFG: WiFi key:       ***
[2018-03-06T22:10:40.389Z] CNFG: WiFi conn_type: 1
[2018-03-06T22:10:40.390Z] CNFG: WiFi sec_type:  4
[2018-03-06T22:10:40.390Z] FLASH: Loading Activation Data
[2018-03-06T22:10:40.392Z] FLASH: Reading 560 bytes from address 8192
[2018-03-06T22:10:40.393Z] FLASH: Loading data succeeded
[2018-03-06T22:10:40.394Z] CNFG: Gateway ID:         arjanvanb-gw-1
[2018-03-06T22:10:40.395Z] CNFG: Gateway Key:        ***
[2018-03-06T22:10:40.396Z] CNFG: Account Server URL: https://account.thethingsnetwork.org
[2018-03-06T22:10:40.396Z] CNFG: Locked:             true
[2018-03-06T22:10:40.397Z] CNFG: Locked first time:  false
[2018-03-06T22:10:40.398Z]
[2018-03-06T22:10:40.398Z] MAIN: Leaving state 1
[2018-03-06T22:10:40.399Z] MAIN: Entering state 2
[2018-03-06T22:10:40.400Z] INET: State change to 0
[2018-03-06T22:10:40.734Z] LORA: Initialisation complete
[2018-03-06T22:10:40.735Z] LORA: Changing state from 0 to 1
[2018-03-06T22:10:47.726Z] LORA: Wait init complete, waiting for application.
[2018-03-06T22:10:47.727Z] LORA: Changing state from 1 to 2
[... 74 lines cut ...]
[2018-03-07T17:03:33.363Z] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-07T17:03:43.338Z] MON: SYS Stack size: 3959
[2018-03-07T17:03:43.339Z] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-07T17:03:53.314Z] MON: SYS Stack size: 3959
[2018-03-07T17:03:53.315Z] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-07T17:04:03.288Z] MON: SYS Stack size: 3959
[2018-03-07T17:04:03.289Z] MON: heap usage: 148KB (148KB), free: 191KB
[2018-03-07T17:04:13.066Z] BTN: change to state: 1
[2018-03-07T17:04:13.248Z] BTN: change to state: 2
[2018-03-07T17:04:13.250Z]
[2018-03-07T17:04:13.251Z] MAIN: L
[2018-03-07T17:04:13.252Z] *** Application reboot
[2018-03-07T17:04:13.300Z]
[2018-03-07T17:04:15.463Z] SNTP: State change from 0 to 0
[2018-03-07T17:04:15.466Z] SNTP: State change from 0 to 0
[2018-03-07T17:04:15.467Z]
[2018-03-07T17:04:15.467Z]
[2018-03-07T17:04:15.468Z]
[2018-03-07T17:04:15.469Z] **************************
[2018-03-07T17:04:15.469Z] *   The Things Network   *
[2018-03-07T17:04:15.470Z] *      G A T E W A Y     *
[2018-03-07T17:04:15.473Z] **************************
```

So, this time it did not get its configuration either. And it was NOT initiated by an MQTT error either. I forget to check which LEDs were on.

avbentem, 2018-03-16:
No changes using @martinichka 's https://github.com/TheThingsProducts/gateway/issues/1#issuecomment-372740116 (so, the latest firmware version, installed using an SD card without using the PIC32 programmer).

Also, again no difference if WiFi or ethernet is used. (And the Raspberry Pi that [is monitoring](https://github.com/avbentem/raspi-uart-monitor) my gateway's serial output, seems to be able to report all network problems using the very same WiFi connection.)

```text
Firmware name: AmazingAckermann, type: 0, version: 1.0.1, commit: 91648b7e, timestamp: 1520959697
Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
Build time: Mar 13 2018 16:49:17
```

avbentem, 2018-04-04:
Problem still exists in 1.0.3 (still loaded from an SD card, while it was officially released today as well, I think). This time initiated by a reboot for a firmware check, while it was running just fine:

```text
[2018-04-04T19:40:26.171Z] LORA: Accepted packet
[2018-04-04T19:40:26.202Z] MQTT: Sending UPLINK OK
[2018-04-04T19:40:26.240Z] LORA: Packet dropped! Bad CRC
[2018-04-04T19:40:32.384Z] MON: SYS Stack size: 2851
[2018-04-04T19:40:32.387Z] MON: heap usage: 263KB (267KB), free: 76KB
[2018-04-04T19:40:42.377Z] MON: SYS Stack size: 2851
[2018-04-04T19:40:42.380Z] MON: heap usage: 263KB (267KB), free: 76KB
[2018-04-04T19:40:52.369Z] MON: SYS Stack size: 2851
[2018-04-04T19:40:52.372Z] MON: heap usage: 263KB (267KB), free: 76KB
[2018-04-04T19:40:54.008Z] 
[2018-04-04T19:40:54.012Z] MAIN: Rebooting gateway for firmware update check
[2018-04-04T19:40:54.801Z] LORA: Accepted packet
[2018-04-04T19:40:56.602Z] SNTP: State change from 7 to 1
[2018-04-04T19:40:57.700Z] SNTP: State change from 1 to 2
[2018-04-04T19:40:58.798Z] SNTP: State change from 2 to 3
[2018-04-04T19:40:59.897Z] SNTP: State change from 3 to 4
[2018-04-04T19:41:00.912Z] LORA: Accepted packet
[2018-04-04T19:41:00.993Z] SNTP: State change from 4 to 5
[2018-04-04T19:41:01.011Z] SNTP: State change from 5 to 6
[2018-04-04T19:41:01.013Z] SNTP: State change from 6 to 7
[2018-04-04T19:41:02.890Z] SNTP: State change from 0 to 0
[2018-04-04T19:41:02.892Z] SNTP: State change from 0 to 0
[2018-04-04T19:41:02.895Z] 
[2018-04-04T19:41:02.895Z] 
[2018-04-04T19:41:02.896Z] 
[2018-04-04T19:41:02.899Z] **************************
[2018-04-04T19:41:02.900Z] *   The Things Network   *
[2018-04-04T19:41:02.903Z] *      G A T E W A Y     *
[2018-04-04T19:41:02.905Z] **************************
[2018-04-04T19:41:02.911Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1521721252
[2018-04-04T19:41:02.939Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-04-04T19:41:02.940Z] Build time: Mar 22 2018 12:21:52
[2018-04-04T19:41:02.954Z] Reboot reason: 0x10
[2018-04-04T19:41:02.963Z] BOOT: (persisted info) 6F 72 72 65 01 10 E3 2C E7 16 05 49 B2 3A 51 B7 
```

...after which it got neither ethernet nor WiFi, until I pressed the Mode button at 2018-04-04T21:38:03.384Z.

bvdbrule, 2018-04-05:
I have received a OTS firmware upgrade v1.0.3-2c56ce09 (2018-04-04T12:54:45Z), but after connecting a serial esp-link solution from BoRRoZ and turning "on"  in the console the switches Automatically update gateway and Beta Updates it goes to version v1.0.3-2c56ce09 from 2018-03-22T12:20:52Z. but after some time it loops: hereby the log 

nging state from 0 to 0

MAIN: Leaving state 0
MAIN: Entering state 1
FLASH: Magic bytes found: wifi config present
FLASH: Magic bytes found: activation data present
FLASH: Magic bytes found: FOTA data present
FLASH: Loading Firmware Data
CNFG: (Firmware HASH (sha256)) 87 93 F2 0A DA 7C FB 56 3C
EB E0 29 21 C5 BB 76 F3 9C A6 A4 A9 2C CE 01 EC A8 29 5B 59
48 9C 20 
FLASH: Loading WiFi Data
CNFG: WiFi SSID:      88888888-Lora
CNFG: WiFi key:       ***
CNFG: WiFi conn_type: 1
CNFG: WiFi sec_type:  4
FLASH: Loading Activation Data
CNFG: Gateway ID:         vdbc-lora-gw
CNFG: Gateway Key:        ***
CNFG: Account Server URL:
https://account.thethingsnetwork.org
CNFG: Locked:             true
CNFG: Locked first time:  false

MAIN: Leaving state 1
MAIN: Entering state 2
INET: State change to 0
LORA: Initialisation complete
LORA: Changing state from 0 to 1
WIFI: Entering state 1
ETH: IP Address: 0.0.0.0 
WIFI: Entering state 4
WIFI: Entering SCAN state 1
SNTP: State change from 0 to 1
Scan is completed successfully
WIFI: Entering SCAN state 2
WIFI: Entering SCAN state 5
WIFI: Entering SCAN state 0
WIFI: Entering state 2
WIFI: Disabling modules
Head magic match void: trying to free an already freed
block, ignore
SNTP: State change from 1 to 1
WIFI: Entering state 3
WIFI: Enabling modules for client
WIFI: Entering state 6

> WIFI: IP Address: 0.0.0.0 
CB: INET: Gateway has WiFi
INET: State change to 2
INET: Connected to a network, waiting for DHCP lease,
checking validity with ping
SNTP: State change from 1 to 2
WIFI: IP Address: 192.168.101.200 
LORA: Wait init complete, waiting for application.
LORA: Changing state from 1 to 2
MON: SYS Stack size: 3967
MON: heap usage: 147KB (156KB), free: 192KB
INET: State change to 3
INET: Ping probe
INET: Error sending probe on Eth
INET: Ping response from MRF24WN, set as default
INET: State change to 4
SNTP: State change from 2 to 3
SNTP: State change from 3 to 4
SNTP: State change from 4 to 5
SNTP: State change from 5 to 6
SNTP: State change from 6 to 7
[... 2562 lines cut ...]
MON: SYS Stack size: 2837
MON: heap usage: 169KB (266KB), free: 169KB
HTTP: TCPIP_DNS_IsResolved returned failure code -5
MQTT: Wrong DNS, stopping operation

MAIN: MQTT error

MAIN: Leaving state 5
LORA: Accepted packet
MAIN: Entering state 6
INET: State change to 0
WIFI: Disabling modules
CB: Disconnect
Head magic match void: trying to free an already freed
block, ignore
SNTP: State change from 6 to 1
WIFI: Entering state 3
WIFI: Enabling modules for client
WIFI: Entering state 6
WIFI: IP Address: 0.0.0.0 
CB: SNTP: State change from 1 to 2
INET: Gateway has WiFi
INET: State change to 2
INET: Connected to a network, waiting for DHCP lease,
checking validity with ping

avbentem, 2018-05-02:
Today's 1.0.4 gets me a reboot loop, so I am back at 1.0.3 which is still giving me this issue a few times per week. (I am using ethernet; have not tested 1.0.4 with WiFi.)

avbentem, 2018-05-02:
@bvdbrule can you please edit your earlier comment to put the long log into something like the following?

    <details>
    <summary>Reboot loop in 1.0.3</summary>
    ```text
    your
    log
    here
    ```
    </details>

Thanks! (I will delete this comment later.)

avbentem, 2018-08-05:
@bvdbrule, did you see my comment? Please edit, many thanks!

avbentem, 2018-08-05:
I've upgraded from 1.0.3 to 1.0.5 on Tuesday July 17th, around 9 PM Amsterdam time. This is using the factory bootloader. The problem still exists since the upgrade, but the cause of the reboot and the logging after the reboot are different. So maybe it's a different problem after all.

Since the upgrade my gateway has run into `*** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking` 225 times; see #52. In 3 of those 225 occurences it failed to recover after it rebooted: it got an ethernet connection, and showed `HTTP: Starting connection` but never showed anything about the network after that (like it did NOT show `Gateway bridging`), though it was receiving LoRa packets. A manual reboot using the pink Mode button fixed this. 

Apart from the following 3 occasions, the gateway has rebooted many, many more times, like after an MQTT error and also for the 222 other occurences of the above assert. But for those reboots, it recovered just fine.

<details>
<summary>Saturday July 21st, no gateway bridging after reboot, until manual reboot, in 1.0.5</summary>

```txt
[2018-07-21T05:07:09.551Z] MQTT: Sending status packet
[2018-07-21T05:07:09.585Z] MQTT: Sending status succeeded: 671
[2018-07-21T05:07:13.413Z] MON: SYS Stack size: 2837
[2018-07-21T05:07:13.416Z] MON: TCPIP Stack size: 3771
[2018-07-21T05:07:13.418Z] MON: APP Stack size: 3271
[2018-07-21T05:07:13.420Z] MON: LoRa Stack size: 3857
[2018-07-21T05:07:13.424Z] MON: heap usage: 282KB (283KB), free: 56KB
[2018-07-21T05:07:17.288Z] LORA: Kic
[2018-07-21T05:07:17.294Z] *** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking
[2018-07-21T05:07:17.297Z] 
[2018-07-21T05:07:19.324Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.328Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.331Z] 
[2018-07-21T05:07:19.332Z] 
[2018-07-21T05:07:19.333Z] 
[2018-07-21T05:07:19.334Z] **************************
[2018-07-21T05:07:19.337Z] *   The Things Network   *
[2018-07-21T05:07:19.339Z] *      G A T E W A Y     *
[2018-07-21T05:07:19.342Z] **************************
[2018-07-21T05:07:19.346Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-07-21T05:07:19.367Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-07-21T05:07:19.369Z] Build time: Jul 17 2018 08:12:51
[2018-07-21T05:07:19.382Z] Reboot reason: 0x40
[2018-07-21T05:07:19.391Z] BOOT: (persisted info) 6F 72 72 65 01 40 AB 2C E7 16 05 C9 B2 3A 50 97 
[2018-07-21T05:07:19.393Z] 
[2018-07-21T05:07:19.394Z] 
[2018-07-21T05:07:19.395Z] 
[2018-07-21T05:07:19.396Z] 
[2018-07-21T05:07:19.397Z] WIFI: Entering state 0
[2018-07-21T05:07:19.399Z] WIFI: Entering SCAN state 0
[2018-07-21T05:07:19.400Z] 
[2018-07-21T05:07:19.401Z] MAIN: Initialisation complete
[2018-07-21T05:07:19.415Z] LORA: Changing state from 0 to 0
[2018-07-21T05:07:19.570Z] 
[2018-07-21T05:07:19.573Z] MAIN: Leaving state 0
[2018-07-21T05:07:19.575Z] MAIN: Entering state 1
[2018-07-21T05:07:19.582Z] FLASH: Magic bytes found: wifi config present
[2018-07-21T05:07:19.589Z] FLASH: Magic bytes found: activation data present
[2018-07-21T05:07:19.593Z] FLASH: Magic bytes found: FOTA data present
[2018-07-21T05:07:19.596Z] FLASH: Loading Firmware Data
[2018-07-21T05:07:19.607Z] CNFG: (Firmware HASH (sha256)) 93 F4 C1 0E 77 81 3F 60 0D A8 A5 97 57 7C DD 9C 95 64 1F DF F4 8D 44 3F 92 9A 20 A9 2D B4 B9 F5 
[2018-07-21T05:07:19.610Z] FLASH: Loading WiFi Data
[2018-07-21T05:07:19.613Z] CNFG: WiFi SSID:      <redacted>
[2018-07-21T05:07:19.616Z] CNFG: WiFi key:       ***
[2018-07-21T05:07:19.618Z] CNFG: WiFi conn_type: 1
[2018-07-21T05:07:19.621Z] CNFG: WiFi sec_type:  4
[2018-07-21T05:07:19.623Z] FLASH: Loading Activation Data
[2018-07-21T05:07:19.626Z] CNFG: Gateway ID:         arjanvanb-gw-1
[2018-07-21T05:07:19.629Z] CNFG: Gateway Key:        ***
[2018-07-21T05:07:19.635Z] CNFG: Account Server URL: https://account.thethingsnetwork.org
[2018-07-21T05:07:19.638Z] CNFG: Locked:             true
[2018-07-21T05:07:19.641Z] CNFG: Locked first time:  false
[2018-07-21T05:07:19.643Z] 
[2018-07-21T05:07:19.645Z] MAIN: Leaving state 1
[2018-07-21T05:07:19.647Z] MAIN: Entering state 2
[2018-07-21T05:07:19.650Z] INET: State change to 0
[2018-07-21T05:07:19.947Z] LORA: Initialisation complete
[2018-07-21T05:07:19.950Z] LORA: Changing state from 0 to 1
[2018-07-21T05:07:20.202Z] WIFI: Entering state 1
[2018-07-21T05:07:20.205Z] ETH: IP Address: 0.0.0.0 
[... 394 lines cut ...]
[2018-08-02T00:38:59.802Z] LGMD:LORA: Accepted packet
...
...
[2018-08-02T04:32:05.113Z] LGMD:Rejected packet (0x11)
[2018-08-02T04:32:05.116Z] 
[2018-08-02T04:32:05.956Z] SNTP: State change from 1 to 2
[2018-08-02T04:32:06.897Z] BTN: change to state: 1
[2018-08-02T04:32:06.998Z] BTN: change to sta
[2018-08-02T04:32:07.001Z] *** Application reboot
[2018-08-02T04:32:07.008Z] 
[2018-08-02T04:32:09.030Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.033Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.035Z] 
[2018-08-02T04:32:09.037Z] 
[2018-08-02T04:32:09.038Z] 
[2018-08-02T04:32:09.040Z] **************************
[2018-08-02T04:32:09.042Z] *   The Things Network   *
[2018-08-02T04:32:09.045Z] *      G A T E W A Y     *
[2018-08-02T04:32:09.047Z] **************************
[2018-08-02T04:32:09.052Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-08-02T04:32:09.074Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-08-02T04:32:09.077Z] Build time: Jul 17 2018 08:12:51
[2018-08-02T04:32:09.091Z] Reboot reason: 0x40
```
</details>

avbentem, 2018-08-05:
I've upgraded from 1.0.3 to 1.0.5 on Tuesday July 17th, around 9 PM Amsterdam time. This is using the factory bootloader. The problem still exists since the upgrade, but the cause of the reboot and the logging after the reboot are different. So maybe it's a different problem after all.

The logging might also indicate this is related to #47, however, now the LoRa module is actually working just fine. So I guess it's related to this very issue, not to #47.

Since the upgrade my gateway has run into `*** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking` 225 times; see #52. In 3 of those 225 occurences it failed to recover after it rebooted: it got an ethernet connection, and showed `HTTP: Starting connection` but never showed anything about the network after that (like it did NOT show `Gateway bridging`), though it was receiving LoRa packets. A manual reboot using the pink Mode button fixed this. 

Apart from the following 3 occasions, the gateway has rebooted many, many more times, like after an MQTT error and also for the 222 other occurences of the above assert. But for those reboots, it recovered just fine.

<details>
<summary>Saturday July 21st, no gateway bridging after reboot, until manual reboot, in 1.0.5</summary>

```txt
[2018-07-21T05:07:09.551Z] MQTT: Sending status packet
[2018-07-21T05:07:09.585Z] MQTT: Sending status succeeded: 671
[2018-07-21T05:07:13.413Z] MON: SYS Stack size: 2837
[2018-07-21T05:07:13.416Z] MON: TCPIP Stack size: 3771
[2018-07-21T05:07:13.418Z] MON: APP Stack size: 3271
[2018-07-21T05:07:13.420Z] MON: LoRa Stack size: 3857
[2018-07-21T05:07:13.424Z] MON: heap usage: 282KB (283KB), free: 56KB
[2018-07-21T05:07:17.288Z] LORA: Kic
[2018-07-21T05:07:17.294Z] *** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking
[2018-07-21T05:07:17.297Z] 
[2018-07-21T05:07:19.324Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.328Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.331Z] 
[2018-07-21T05:07:19.332Z] 
[2018-07-21T05:07:19.333Z] 
[2018-07-21T05:07:19.334Z] **************************
[2018-07-21T05:07:19.337Z] *   The Things Network   *
[2018-07-21T05:07:19.339Z] *      G A T E W A Y     *
[2018-07-21T05:07:19.342Z] **************************
[2018-07-21T05:07:19.346Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-07-21T05:07:19.367Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-07-21T05:07:19.369Z] Build time: Jul 17 2018 08:12:51
[2018-07-21T05:07:19.382Z] Reboot reason: 0x40
[2018-07-21T05:07:19.391Z] BOOT: (persisted info) 6F 72 72 65 01 40 AB 2C E7 16 05 C9 B2 3A 50 97 
[2018-07-21T05:07:19.393Z] 
[2018-07-21T05:07:19.394Z] 
[2018-07-21T05:07:19.395Z] 
[2018-07-21T05:07:19.396Z] 
[2018-07-21T05:07:19.397Z] WIFI: Entering state 0
[2018-07-21T05:07:19.399Z] WIFI: Entering SCAN state 0
[2018-07-21T05:07:19.400Z] 
[2018-07-21T05:07:19.401Z] MAIN: Initialisation complete
[2018-07-21T05:07:19.415Z] LORA: Changing state from 0 to 0
[2018-07-21T05:07:19.570Z] 
[2018-07-21T05:07:19.573Z] MAIN: Leaving state 0
[2018-07-21T05:07:19.575Z] MAIN: Entering state 1
[2018-07-21T05:07:19.582Z] FLASH: Magic bytes found: wifi config present
[2018-07-21T05:07:19.589Z] FLASH: Magic bytes found: activation data present
[2018-07-21T05:07:19.593Z] FLASH: Magic bytes found: FOTA data present
[2018-07-21T05:07:19.596Z] FLASH: Loading Firmware Data
[2018-07-21T05:07:19.607Z] CNFG: (Firmware HASH (sha256)) 93 F4 C1 0E 77 81 3F 60 0D A8 A5 97 57 7C DD 9C 95 64 1F DF F4 8D 44 3F 92 9A 20 A9 2D B4 B9 F5 
[2018-07-21T05:07:19.610Z] FLASH: Loading WiFi Data
[2018-07-21T05:07:19.613Z] CNFG: WiFi SSID:      <redacted>
[2018-07-21T05:07:19.616Z] CNFG: WiFi key:       ***
[2018-07-21T05:07:19.618Z] CNFG: WiFi conn_type: 1
[2018-07-21T05:07:19.621Z] CNFG: WiFi sec_type:  4
[2018-07-21T05:07:19.623Z] FLASH: Loading Activation Data
[2018-07-21T05:07:19.626Z] CNFG: Gateway ID:         arjanvanb-gw-1
[2018-07-21T05:07:19.629Z] CNFG: Gateway Key:        ***
[2018-07-21T05:07:19.635Z] CNFG: Account Server URL: https://account.thethingsnetwork.org
[2018-07-21T05:07:19.638Z] CNFG: Locked:             true
[2018-07-21T05:07:19.641Z] CNFG: Locked first time:  false
[2018-07-21T05:07:19.643Z] 
[2018-07-21T05:07:19.645Z] MAIN: Leaving state 1
[2018-07-21T05:07:19.647Z] MAIN: Entering state 2
[2018-07-21T05:07:19.650Z] INET: State change to 0
[2018-07-21T05:07:19.947Z] LORA: Initialisation complete
[2018-07-21T05:07:19.950Z] LORA: Changing state from 0 to 1
[... 396 lines cut ...]
[2018-08-02T00:38:59.802Z] LGMD:LORA: Accepted packet
...
...
[2018-08-02T04:32:05.113Z] LGMD:Rejected packet (0x11)
[2018-08-02T04:32:05.116Z] 
[2018-08-02T04:32:05.956Z] SNTP: State change from 1 to 2
[2018-08-02T04:32:06.897Z] BTN: change to state: 1
[2018-08-02T04:32:06.998Z] BTN: change to sta
[2018-08-02T04:32:07.001Z] *** Application reboot
[2018-08-02T04:32:07.008Z] 
[2018-08-02T04:32:09.030Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.033Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.035Z] 
[2018-08-02T04:32:09.037Z] 
[2018-08-02T04:32:09.038Z] 
[2018-08-02T04:32:09.040Z] **************************
[2018-08-02T04:32:09.042Z] *   The Things Network   *
[2018-08-02T04:32:09.045Z] *      G A T E W A Y     *
[2018-08-02T04:32:09.047Z] **************************
[2018-08-02T04:32:09.052Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-08-02T04:32:09.074Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-08-02T04:32:09.077Z] Build time: Jul 17 2018 08:12:51
[2018-08-02T04:32:09.091Z] Reboot reason: 0x40
```
</details>