TheThingsProducts/gateway #10: Unable to turn off wifi AP mode on the gateway

sourceperl, 2018-02-14:
Hi,

I use the ethernet link to connect the gateway to the internet. 

After activation the wifi AP mode is not turn off and my gateway always broadcast the SSID ""Things-Gateway-0001ECxxxxxx". The default password "thethings" also remain active... It's really not secure.

I'm unable to fix  this. I try the embedded web manager, the serial debug (the wifi cmd seem not work) and https://console.thethingsnetwork.org/gateways.

My firmware is :
(Firmware HASH (sha256)) 69 AE B7 78 1F 49 4E 7F BC B6 C7 CD 9C 59 4F 5D FA AA 3D 81 D4 9C 56 90 A6 83 81 98 FF 18 88 6A

Bootloader is:
Firmware name: AmazingAckermann, type: 0, version: 1.0.0, commit: 917719b9, timestamp: 1498499973
Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
Build time: Jun 26 2017 19:59:53

sourceperl, 2018-02-15:
after some test, i find the command "if wlan0 down" this turn off wifi access point but it's not persistent after reboot.

IJeffray, 2018-02-20:
I would very much like to run with Ethernet only and the WLAN module completely disabled.  For security, and for keeping the air clear.   I'm very surprised this isn't an option already as it would surely be simple to implement.   It could even be 'automatic' - if Ethernet is present at bootup, disable the Wifi -- no need for any user interface.

cyberjunky, 2018-02-24:
Sometimes you might need the wifi to be able to reconfigure/activate it, maybe it's an idea to toggle wifi on off by using a file flag on the sd card, so when a user creates a file on the sd card named `'nowifi' `disable it.

wintermi, 2018-03-15:
Just got my TTN Gateway and have the same issue.  I only want to use the Ethernet connection and the only way I have found at the moment to remove the Wifi AP is to actually connect the Wifi.  Not a good solution as now both Wifi & Ethernet is configured.

closed 2018-04-21 by KrishnaIyer

lanmarc77, 2019-02-18:
I am not sure if I understood this thread. But I still see the SSID being broadcasted in the current firmware even though I have working ethernet connection and /info also only displays Ethernet as connection type.
Inserting the following code in app_wifi.c at line 354 finally stops the broadcast and also saves 0.3W of power:

`_activeConfig = CNFG_NONE;`
`_event=EVENT_NONE;`
`return;`

Who also wants to disable the built in http server comments line 739 in system_config.h:

`//#define TCPIP_STACK_USE_HTTP_SERVER`

hagenbuch, 2019-02-28:
Does this mean I have to compile and install the complete firmware for the TTN gateway?

If yes, I'd rather wait until the first scandal shows up where some attacker exploited a company successfully and came in through their open Things Gateway AP /s

lanmarc77, 2019-02-28:
Yes it does mean compiling and installing the firmware yourself. If you do have the .hex file you only would need an SD card to actually update the gateway. Also compiling is not really more than installing the freely availble IDEs linked in the readme of this repository and cloning this repository. As I already did the detective work how to minimally change the code to achieve the given features this can be done in under an hour easily.
And also yes you are free to wait until something (might) happen but then I do not see how this helps in any way to add features WE want. I so value the possibility to have an open device, knowing I do not get an SLA but be able to do something with it myself.

KrishnaIyer, 2019-03-06:
@lanmarc77: what is the version of your firmware?

lanmarc77, 2019-03-06:
@KrishnaIyer The base of the mentioned modification was v1.0.5 which is the current version of the firmware since July 2018

MartinNPN, 2019-05-14:
@lanmarc77 thanks for the easy solution :)

devZer0, 2023-05-16:
this ticket is closed - but how can i disable wifi now ?

what do i need to do ?

PR #38 (2018-04-21): Fix/keep existing wifi config
Copy of commit message:
```
The previous WiFi patch that restricted AP mode to only when there is no WiFi config causes issues with gateways with exisiting WiFi config.
Though the two are not directly related, the patch seems to have uncovered an existing problem.
In the current code, the WiFi ping timeout is 8s (4x2 tries). But practically, it takes 10-15s to re-connect. So, after 8s, we go to WAIT_FOR_NETWORK state where the WiFi module is restarted.
This causes a loop since everytime the pings timeout at 8s and restart the WiFi module.
By increasing the ping ti [... cut]