TheThingsProducts/gateway #14: Docker Build - fails

grahamehorner, 2018-02-15:
The docker build is failing with the following error

standard_init_linux.go:195: exec user process caused "no such file or directory"

cliffalbert, 2018-02-16:
Docker builds work fine here, might want to post some more output from the build process it self.

grahamehorner, 2018-02-16:
will do I’m wondering if this is a docker on Windows issue? has this  even tried on Windows ?

grahamehorner, 2018-02-21:
more info:

```
Microsoft Windows [Version 10.0.16299.125]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Users\...\source\repos\ttn-gateway\firmware>docker build -t firmware-builder -f Dockerfile.build .
Sending build context to Docker daemon  29.14MB
Step 1/12 : FROM ubuntu:xenial
xenial: Pulling from library/ubuntu
1be7f2b886e8: Pull complete
6fbc4a21b806: Pull complete
c71a6f8e1378: Pull complete
4be3072e5a37: Pull complete
06c6d2f59700: Pull complete
Digest: sha256:e27e9d7f7f28d67aa9e2d7540bdc2b33254b452ee8e60f388875e5b7d9b2b696
Status: Downloaded newer image for ubuntu:xenial
 ---> 0458a4468cbc
Step 2/12 : RUN dpkg --add-architecture i386 &&   apt-get update &&   apt-get install -y build-essential curl sudo git &&   apt-get install -y libc6:i386 libx11-6:i386 libxext6:i386 libstdc++6:i386 libexpat1:i386 &&   rm -rf /var/lib/apt/lists/*
 ---> Running in 9cbe99787891
Get:1 http://archive.ubuntu.com/ubuntu xenial InRelease [247 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]
Get:3 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]
Get:4 http://archive.ubuntu.com/ubuntu xenial/universe Sources [9802 kB]
Get:5 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]
Get:6 http://security.ubuntu.com/ubuntu xenial-security/universe Sources [72.8 kB]
Get:7 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [576 kB]
Get:8 http://security.ubuntu.com/ubuntu xenial-security/main i386 Packages [521 kB]
Get:9 http://security.ubuntu.com/ubuntu xenial-security/restricted amd64 Packages [12.7 kB]
Get:10 http://security.ubuntu.com/ubuntu xenial-security/restricted i386 Packages [12.7 kB]
Get:11 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [400 kB]
Get:12 http://security.ubuntu.com/ubuntu xenial-security/universe i386 Packages [349 kB]
Get:13 http://security.ubuntu.com/ubuntu xenial-security/multiverse amd64 Packages [3486 B]
Get:14 http://security.ubuntu.com/ubuntu xenial-security/multiverse i386 Packages [3675 B]
Get:15 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages [1558 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial/main i386 Packages [1552 kB]
Get:17 http://archive.ubuntu.com/ubuntu xenial/restricted amd64 Packages [14.1 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial/restricted i386 Packages [14.5 kB]
Get:19 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [9827 kB]
Get:20 http://archive.ubuntu.com/ubuntu xenial/universe i386 Packages [9804 kB]
Get:21 http://archive.ubuntu.com/ubuntu xenial/multiverse amd64 Packages [176 kB]
Get:22 http://archive.ubuntu.com/ubuntu xenial/multiverse i386 Packages [172 kB]
Get:23 http://archive.ubuntu.com/ubuntu xenial-updates/universe Sources [240 kB]
Get:24 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [942 kB]
Get:25 http://archive.ubuntu.com/ubuntu xenial-updates/main i386 Packages [877 kB]
Get:26 http://archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 Packages [13.1 kB]
Get:27 http://archive.ubuntu.com/ubuntu xenial-updates/restricted i386 Packages [13.1 kB]
Get:28 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [757 kB]
Get:29 http://archive.ubuntu.com/ubuntu xenial-updates/universe i386 Packages [701 kB]
Get:30 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 Packages [18.5 kB]
Get:31 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse i386 Packages [17.3 kB]
Get:32 http://archive.ubuntu.com/ubuntu xenial-backports/main amd64 Packages [5153 B]
Get:33 http://archive.ubuntu.com/ubuntu xenial-backports/main i386 Packages [5141 B]
Get:34 http://archive.ubuntu.com/ubuntu xenial-backports/universe amd64 Packages [7172 B]
Get:35 http://archive.ubuntu.com/ubuntu xenial-backports/universe i386 Packages [7170 B]
Fetched 39.0 MB in 19s (2011 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils bzip2 ca-certificates cpp cpp-5 dpkg-dev fakeroot g++ g++-5 gcc
  gcc-5 gcc-5-base git-man ifupdown iproute2 isc-dhcp-client isc-dhcp-common
  krb5-locales less libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libasan2 libasn1-8-heimdal libatm1 libatomic1
  libbsd0 libc-dev-bin libc6-dev libcc1-0 libcilkrts5 libcurl3-gnutls
  libdns-export162 libdpkg-perl libedit2 liberror-perl libexpat1 libfakeroot
  libffi6 libfile-fcntllock-perl libgcc-5-dev libgdbm3 libgmp10 libgnutls30
  libgomp1 libgssapi-krb5-2 libgssapi3-heimdal libhcrypto4-heimdal
  libheimbase1-heimdal libheimntlm0-heimdal libhogweed4 libhx509-5-heimdal
[... 744 lines cut ...]
64 Bit, check libraries
Check for 32 Bit libraries
Verifying archive integrity... All good.
Uncompressing MPLAB X v3.45 Installer....
The udev rules have been updated.  Some distributions require that you re-start your computer for the changes to take effect otherwise USB communications and/or hotplug will not work.
Removing intermediate container 407dbd9cb0c1
 ---> 99badf0be6b7
Step 10/12 : RUN ln -s /opt/microchip/mplabx/v${mplab_version}/mplab_ide /opt/microchip/mplab_ide
 ---> Running in fadd993ef1af
Removing intermediate container fadd993ef1af
 ---> 7884c7f86d7a
Step 11/12 : RUN /tmp/xc32-v${xc32_version}-full-install-linux-installer.run -- --mode unattended --netservername ""
 ---> Running in 1f9a85f7788f
Removing intermediate container 1f9a85f7788f
 ---> 0b93d77ec23b
[... cut ...] -mode unattended
 ---> Running in 3b309dcec7f5
Removing intermediate container 3b309dcec7f5
 ---> 0a806c1a528f
Successfully built 0a806c1a528f
Successfully tagged firmware-builder:latest
SECURITY WARNING: You are building a Docker image from Windows against a non-Windows Docker host. All files and directories added to build context will have '-rwxr-xr-x' permissions. It is recommended to double check and reset permissions for sensitive files and directories.
```

above is the docker build output

grahamehorner, 2018-02-21:
```
C:\Users\...\source\repos\ttn-gateway\firmware>docker run --rm -it -v C:\Users\...\source\repos\ttn-gateway\firmware:/build -w /build firmware-builder ./compile.sh ./generate_hex_with_checksum.s
**standard_init_linux.go:195: exec user process caused "no such file or directory"**
```

htdvisser, 2018-02-21:
From a quick Google search I think that it may have something to do with line endings:
https://forums.docker.com/t/standard-init-linux-go-175-exec-user-process-caused-no-such-file/20025
https://stackoverflow.com/questions/46867754/getting-exec-user-process-caused-no-such-file-or-directory-error-when-trying