TheThingsProducts/gateway #18: Not connecting to Network / Gateway Card not defined

samuel-puschacher, 2018-03-08:
Hello, 
I can't get my Gateway connecting to TTN
I get stuck at Step 4.
When I open the Gateway Info Page I get the following:
[image]

When I refresh the Page, the LoRa Config is lost, the Concentrator not defined.
[image]

I tried WiFi & Ethernet in many Combinations and different Networks. Also reset the Gateway, by pressing the Button during Power up.

Pinkfoxi, 2018-03-08:
Hello Samuel

Check the number on the board (you see them on the LoRa-Board-Connector if you remove the white cover - it is only pressed on the case). If number 1636D4, I asume it won't work. I had the same problem with my first gateway. The second 1708D4 works.

Best regards
Werner

Von: samuel-puschacher
Gesendet: Donnerstag, 8. März, 17:51
Betreff: [TheThingsProducts/gateway] Not connecting to Network / Gateway Card not defined (#18)
An: TheThingsProducts/gateway
Cc: Subscribed

Hello,
I can't get my Gateway connecting to TTN
I get stuck at Step 4.
When I open the Gateway Info Page I get the following:
When I refresh the Page, the LoRa Config is lost, the Concentrator not defined.
I tried WiFi & Ethernet in many Combinations and different Networks. Also reset the Gateway, by pressing the Button during Power up.

samuel-puschacher, 2018-03-08:
Hello Werner, 
I checked that already, it is a 1708D4
Best regards,
Samuel

Pinkfoxi, 2018-03-08:
Take another gatewayname like abc-gateway123. Then try to switch off your wlan and to connect only by cable. Try to reset the gateway, hold the reset button on the board over 5 seconds. Should reboot.
If you have a laptop connect them to fix lan (with connection to internet!) and than connect Things-Gateway-xyz... with password thethings by wlan, additional from laptop to gateway.  It use normaly less then 60 seconds to register. Works if 4 Leds are green. Good luck.

If this also not work, I asume you have a bad one, like my first gateway.

Best regards.
Werner

[image 1]<https://user-images.githubusercontent.com/9848082/37163688-ac90606c-22f8-11e8-93ad-81bfc6646770.png>

[image 2]<https://user-images.githubusercontent.com/9848082/37163886-34cee688-22f9-11e8-99fb-a1be218b3af6.png>

Pinkfoxi, 2018-03-08:
See also here, there are some people with troubles:
https://www.thethingsnetwork.org/forum/t/ttn-gateway-central/10976/629

samuel-puschacher, 2018-03-08:
Yes, I saw that, already
I tried three different unique Names, A few reses, only WiFi, only LAN, and the combination..... 
Everything I could think of......
Best regards,
Samuel

Pinkfoxi, 2018-03-08:
https://www.thethingsnetwork.org/docs/gateways/gateway/ this is the best way to configure your gateway. You need no own WiFi-Network! Try it by wire, use only the ThingsNetwork-Gateway-WiFi to configure the new gateway. If the gateway works, all 4 Led's active, you can start your wlan and change the Wifi-Address of the gateay.

samuel-puschacher, 2018-03-08:
That's exactly, what I did
Now also tried to update the firmware over microSD Card to the Current stable Version from Github. But this results in the same behavior.

grahamehorner, 2018-03-12:
@samuel-puschacher unfortunately a lot of people are seeing these issues and coming up with possibly incorrect conclusions about the root cause and/or weather the cause is firmware or hardware or a mixture of both.  As I understand it @Pinkfoxi is incorrect in so much that that socket for the card isn’t the issue as other have swapped working cards placing these into faulty gateways which resolved the issue.

I’m too seeing the same issue as you have been with a DOA gateway and have been trying to move this forward for months, and believe me @Pinkfoxi if it were as simple as returning the faulty gateway and getting a replacement; I think many of use would have done this, but unless your aware of something I’m not please let use know how this can be done?

At this time I’m afraid that I’ve simply written off the TTN gateway as a bad purchase and it is collecting dust in the office as it’s not fit for anything due to these very frustrating issues and lack of regular communication on progress in resolving

ReiniervdL, 2018-03-15:
@samuel-puschacher Were you able to install the new firmware on your TTN gateway? In my case, the firmware on SD only loads after I have inserted the LoRa card from a working gateway. With the "bad" LoRa card, I get the same loop: reset after failure to configure the LoRa module.

samuel-puschacher, 2018-03-15:
@ReiniervdL YesI was able to update and connect the gateway
But I'm still watching an Issue of a delay, sometimes messages get delivered to TTN after a few minutes

closed 2018-03-15 by samuel-puschacher