TheThingsProducts/gateway #22: Gateway hangs intermittently

j4zzcat, 2018-03-22:
Hi, 

My gateway stops working intermittently. When this happens, the leds indicate a wifi problem (led1:on, led2:flashing_fast, led3:on, led4:off, led5:off), and a power off/on is needed. Needless to say, my wifi network is stable as a rock. The problem usually occurs within an hour of operation, and had happened dozens of time since I initially installed the gateway, a few days ago.

With **the white lid open**, the problem had not manifested itself for 24 hours, after which I ended my test. I'm therefor speculating that the problem is somehow related to the ambient temperature inside the gateway enclosure. 

```
Hardware: v1
Bootloader: r1-7167873a (2017-06-02T13:48:18Z)
Firmware: v1.0.2-475aef56 (2018-03-15T09:43:08Z)
```

j4zzcat, 2018-03-26:
Script started on Mon Mar 26 08:50:36 2018
command: picocom /dev/cu.usbmodem144331 -b 115200
picocom v3.1

port is        : /dev/cu.usbmodem144331
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

!! Settings mismatch !! Type [C-a] [C-v] to see actual port settings

Type [C-a] [C-h] to see available commands
Terminal ready
AltSoftSerial Test Begin
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
LORA: Accepted packet
MQTT: Sending UPLINK OK
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MQTT: Sending status packet
MQTT: Sending status succeeded: 37
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
LORA: Packet dropped! Bad CRC
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MQTT: Sending status packet
MQTT: Sending status succeeded: 38
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
LORA: Accepted packet
MQTT: Sending UPLINK OK
LORA: Accepted packet
MQTT: Pending UPLINK OK
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
MQTT: Sending status packet
MQT\: Sending status succeeded: 39
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
SNTP: State change fro[º¢½1
SNTP: Statd change from 1 to 2
MON: SYS Stack size: 2847
MON: heap usage: 262KB (263KB), free: 77KB
SNTP: State change from 2 to 3
SNTP: State change from 3 to 4
[... 204 lines cut ...]
GIFI: IP Address: 192.168.84.1 
CB: MON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
LORA: Packet dropped! Bad CRC
MON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
MON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
LORA: Accepted packet
MON: SYS Stack size: 2847
OON: heap us`ge: 167KB (263KB), free: 171KB
LORA: Accepted packet
MON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
OON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
MON: SYS Stack size: 2847
MON: heap usage: 167KB (263KB), free: 171KB
MON: SYS Stack siza: 2847
MOÎ: heap usage: 167KB (263KB), free: 071KB

Terminating...
Thanks for using picocom

Script done on Mon Mar 26 09:01:36 2018

htdvisser, 2018-03-26:
Duplicate of #21 (except that in this case the gateway didn't reboot)

closed 2018-03-26 by htdvisser

j4zzcat, 2018-03-26:
After additional testing, there’s no evidence that the problem relates to the ambient temperature within the gateway enclosure.