TheThingsProducts/gateway #27: Shell access to The Things Gateway (recently bought)

petenorth, 2018-03-30:
This is a **question**

### What do you want to do? What do you expect?

I would like to be able to use the ttn gateway  to connect to other (private) networks  (this is 'as well as' i.e. for prototyping purposes, I still intend to have my gateway connected to TTN most of the time :-) 

### What steps did you take? (Give as many details as possible so that we can reproduce it)

I configured the gateway using the conole based registration process where the list of routers is restricted to those that are part of the The Things Network

### What went wrong? What is missing?

An option to configure the gateway to use different router address and port (presumably the port it uses is 1883 rather than the UDP server port of the router 1700)

### What firmware is running on your gateway?

Not sure - how do I find this?

### What do the logs show?

How do I view the logs? Is it possible to logon to the OS of the gateway? SSH into it? 

If there was a way of SSH onto the gateway then presumably a configuration file could be edited with a different router and port address? I am presuming that it is running the gateway bridge and therefore only communicates using gRPC over MQTT. 

### Any Screenshots or Photos?

N/A

### Do you think you can try to fix this yourself and submit a pull request?

Not without some help explaining how to open a shell in the device.

KrishnaIyer, 2018-03-30:
Hi @petenorth: 
The Things Gateway does not run a linux based OS so there is no shell. 

> I configured the gateway using the conole based registration process where the list of routers is restricted to those that are part of the The Things Network

This is not true. You can enter any plaintext into the field and it will accept it. So entering an IP address such as `192.168.1.5` should set the `MQTT` address to `192.168.1.5:1883`. This would then be where your uplinks are published to. 
If you set a private network server on the same IP, this should technically redirect the uplinks to that network server
  
> How do I view the logs? Is it possible to logon to the OS of the gateway? SSH into it?

You need to use an FTDI with USB to do this. These are _read-only debug_ logs:
https://www.thethingsnetwork.org/docs/gateways/gateway/faq.html#q-i-want-to-get-in-depth-insightread-debug-messages-of-my-gateway-is-that-possible

>  I am presuming that it is running the gateway bridge and therefore only communicates using gRPC over MQTT.

The communication between the gateway and the backend is only via `MQTT`. There is no gRPC involved. 

Good luck and let us know how it goes.

petenorth, 2018-03-30:
Thanks Krishna for the thorough answer I'll update this issue with what I discover!

petenorth, 2018-03-30:
@KrishnaIyer Just an update, the TTN web console doesn't accept a IP address such as you described.

KrishnaIyer, 2018-03-30:
@petenorth : Yes it does. I tried it before answering this post.
[image]

petenorth, 2018-03-30:
[image]

petenorth, 2018-03-30:
OK have worked out why the different behaviour!

When creating a gateway via  'register gateway' the router field can be updated as you say.

For an existing router the screen does not allow the router field to be updated to an IP address it seems to have extra validation.

Thanks,

Peter.

KrishnaIyer, 2018-03-30:
You can still do it using ‘ttnctl’. Erase the existing setting on the gateway and de-activate it.

KrishnaIyer, 2018-03-30:
*re-activate it

KrishnaIyer, 2018-06-07:
@petenorth : After some checks, this cannot be done in our V2 stack as it requires updates to our Account Server. We have added this to our internal features list in V3. Closing this issue for now.

closed 2018-06-07 by KrishnaIyer

khushboo-sah, 2019-12-07:
I am trying to create a gateway but it is saying gateway already created. yes, I did create a gateway with the same pycom board but with a different id. please help me what to do now?