TheThingsProducts/gateway #29: LoRa module fails receiving packets after unexpected reboot, all fine again after manual reboot

avbentem, 2018-04-02:
This is a **bug report**.

### What do you want to do? What do you expect?

After an (unexpected) reboot the gateway's LoRa module should start receiving LoRa packets.

### What went wrong? What is missing?

- After running version 1.0.3 fine for 7 hours, the gateway ran into `FAILMQTT: Sending status failed` and `MAIN: MQTT error`, and rebooted.
- It then showed a lot of `LORA: UART WRITE ERROR!` errors, but eventually reported `LORA: Configuration succeeded` and `LORA: Starting operation`.
- (It also showed a single `LORA: UART TIMEOUT` before `LORA: Configuration succeeded`, but that is also seen in cases where the LoRa module truly starts just fine, and was also seen in version 1.0.1.)
- Despite the `LORA: Starting operation` it was not receiving any LoRa packets at all. (It showed _no_ `LORA: Accepted packet`, and neither any `LORA: Accepted packet` nor `LORA: Packet dropped! Bad CRC`.)
- After 4 hours, after pressing the Mode button to force another reboot, everything immediately worked just fine again.

The gateway is using ethernet. It seems the same LEDs were on before and after the manual reboot. An LMiC test node is sending many packets on alternating frequencies and spreading factors. That test node was not touched at all when pressing the Mode button, while its packets were received just fine again after that manual intervention. So surely packets should have been received before the manual reboot was triggered.

### What firmware is running on your gateway?

Version 1.0.3 beta channel, loaded using an SD card, using the factory bootloader. 

As an aside:

- I've not used version 1.0.2, as that got me a reboot loop: https://github.com/TheThingsProducts/gateway/issues/1#issuecomment-377886224. 
- Version 1.0.1 also got me one or two occasions where I'd not get any uplinks in TTN Console, but I'd have to dive into old log files to see if at that point no LoRa packets were received (like now), or if this was caused by some MQTT issue.

### What do the logs show?

After an expected reboot today 08:49 UTC (which upgraded from 1.0.1 to 1.0.3), all was fine until a `FAILMQTT: Sending status failed` at 15:50 UTC (17:50 Amsterdam time; all timestamps added by a Raspberry Pi that monitors the serial output):

```text
[2018-04-02T15:48:49.456Z] LORA: Packet dropped! Bad CRC
[2018-04-02T15:48:49.674Z] LORA: Accepted packet
[2018-04-02T15:48:49.706Z] MQTT: Sending UPLINK OK
[2018-04-02T15:48:55.046Z] MQTT: Sending status packet
[2018-04-02T15:48:55.077Z] MQTT: Sending status succeeded: 836
...
[2018-04-02T15:50:24.958Z] MQTT: Sending status packet
[2018-04-02T15:50:27.457Z] FAILMQTT: Sending status failed
[2018-04-02T15:50:27.464Z]
[2018-04-02T15:50:27.478Z] MAIN: MQTT error
[2018-04-02T15:50:27.485Z]
[2018-04-02T15:50:27.486Z] MAIN: Leaving state 5
[2018-04-02T15:50:36.345Z] SNTP: State change from 0 to 0
[2018-04-02T15:50:36.348Z] SNTP: State change from 0 to 0
[2018-04-02T15:50:36.350Z]
[2018-04-02T15:50:36.350Z]
[2018-04-02T15:50:36.352Z]
[2018-04-02T15:50:36.354Z] **************************
[2018-04-02T15:50:36.355Z] *   The Things Network   *
[2018-04-02T15:50:36.357Z] *      G A T E W A Y     *
[2018-04-02T15:50:36.359Z] **************************
[2018-04-02T15:50:36.368Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1521721252
[2018-04-02T15:50:36.387Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-04-02T15:50:36.399Z] Build time: Mar 22 2018 12:21:52
[2018-04-02T15:50:36.401Z] Reboot reason: 0x10
```

After this reboot it got a network connection, and got as far as:

```text
[2018-04-02T15:51:05.179Z] CNFG: Load online user config state change to 7
[2018-04-02T15:51:05.180Z]
[2018-04-02T15:51:05.182Z] CNFG: Configuring LoRa module
[2018-04-02T15:51:05.184Z] LORA: Changing state from 4 to 6
[2018-04-02T15:51:05.187Z] LORA: Starting reconfiguration
[2018-04-02T15:51:05.194Z] LORA: UART WRITE ERROR!
[2018-04-02T15:51:05.201Z] LORA: UART WRITE ERROR!
[2018-04-02T15:51:05.238Z] LORA: version: 00
[2018-04-02T15:51:05.239Z] LORA: UART WRITE ERROR!
[2018-04-02T15:51:05.245Z] LORA: UART WRITE ERROR!
[... 119 lines cut ...]
[2018-04-02T15:53:06.172Z] MON: heap usage: 264KB (266KB), free: 74KB
[2018-04-02T15:53:13.131Z] MQTT: Sending status packet
[2018-04-02T15:53:13.162Z] MQTT: Sending status succeeded: 2
[2018-04-02T15:53:16.156Z] MON: SYS Stack size: 2837
[2018-04-02T15:53:16.159Z] MON: heap usage: 263KB (266KB), free: 76KB
```

...and every now and then also:

```text
[2018-04-02T16:00:58.349Z] SNTP: State change from 7 to 1
[2018-04-02T16:00:59.444Z] SNTP: State change from 1 to 2
[2018-04-02T16:01:00.541Z] SNTP: State change from 2 to 3
[2018-04-02T16:01:01.638Z] SNTP: State change from 3 to 4
[2018-04-02T16:01:02.737Z] SNTP: State change from 4 to 5
[2018-04-02T16:01:02.764Z] SNTP: State change from 5 to 6
[2018-04-02T16:01:02.767Z] SNTP: State change from 6 to 7
```

Pressing the mode button to reset the gateway fixes all this.

### Any Screenshots or Photos?

- [Full logs](https://github.com/TheThingsProducts/gateway/files/1869224/arjan-gateway-20180402.log).
- [LEDs before the manual reboot](https://user-images.githubusercontent.com/1497002/38215826-4754bfba-36c9-11e8-8099-fa5f88d660a9.jpg).

willem4ever, 2018-04-04:
I upgraded this afternoon to `Firmware name: AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1522846485` and see LORA errors, after the first reboot a time-out (never seen this before) not an improvement, actually force as before ....

```
2018-04-04 17:45:45.375022 LORA: UART TIMEOUT
2018-04-04 17:45:45.375061 LORA: Configuration succeeded
2018-04-04 17:45:45.375065 LORA: Starting operation
2018-04-04 17:45:45.375068 LORA: Changing state from 6 to 8
2018-04-04 17:45:51.436553 LORA: Changing state from 8 to 5
2018-04-04 17:45:51.436590 LORA: GOING ASYNC
2018-04-04 17:45:51.436594 LORA: Changing state from 5 to 11
2018-04-04 17:46:51.734110 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:47:52.067410 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:48:52.414499 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:49:52.769109 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:50:53.112584 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:51:53.474841 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:52:53.840829 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:53:54.176918 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:54:54.528266 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:55:54.879097 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:56:55.245150 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:57:55.618957 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:58:55.996203 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 17:59:56.358381 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 18:00:56.732491 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 18:01:57.104779 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 18:02:57.456434 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 18:03:57.816745 LORA: Kick LoRa module with ACK after not acked it for 60s
2018-04-04 18:04:58.188847 LORA: Kick LoRa module with ACK after not acked it for 60s
```

avbentem, 2018-05-12:
Today things were not resolved by pressing the Mode button. Instead, it needed a power cycle, after which it started to receive LoRa packets again. Again, the gateway was running just fine until the failure was initiated by a firmware upgrade check. I did not check the LEDs on the board. It's running 1.0.3 (commit: 2c56ce0, timestamp: 1521721252, Build time: Mar 22 2018 12:21:52) from an SD card, [as 1.0.4 gives me a reboot loop](https://github.com/TheThingsProducts/gateway/issues/1#issuecomment-386065867).

Full logs:
<details>
<summary>Running fine, then no LoRa after reboot for firmware upgrade check</summary>

```text
[2018-05-12T08:50:38.136Z] [info] LORA: Accepted packet
[2018-05-12T08:50:38.167Z] [info] MQTT: Sending UPLINK OK
[2018-05-12T08:50:46.551Z] [info] LORA: Accepted packet
[2018-05-12T08:50:46.579Z] [info] MQTT: Sending UPLINK OK
[2018-05-12T08:50:56.479Z] [error] MAIN: Rebooting gateway for firmware update check
[2018-05-12T08:51:02.875Z] [info] LORA: Accepted packet
[2018-05-12T08:51:05.362Z] [info] SNTP: State change from 0 to 0
[2018-05-12T08:51:05.365Z] [info] SNTP: State change from 0 to 0
[2018-05-12T08:51:05.370Z] [info] **************************
[2018-05-12T08:51:05.372Z] [info] *   The Things Network   *
[2018-05-12T08:51:05.374Z] [info] *      G A T E W A Y     *
[2018-05-12T08:51:05.376Z] [info] **************************
[2018-05-12T08:51:05.384Z] [error] Firmware name: AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1521721252
[2018-05-12T08:51:05.419Z] [info] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-05-12T08:51:05.420Z] [info] Build time: Mar 22 2018 12:21:52
[2018-05-12T08:51:05.421Z] [error] Reboot reason: 0x10
[2018-05-12T08:51:05.449Z] [info] BOOT: (persisted info) 6F 72 72 65 01 10 83 2C E7 16 0D 49 B2 1A 54 B6
[2018-05-12T08:51:05.452Z] [info] WIFI: Entering state 0
[2018-05-12T08:51:05.453Z] [info] WIFI: Entering SCAN state 0
[2018-05-12T08:51:05.455Z] [info] MAIN: Initialisation complete
[2018-05-12T08:51:05.456Z] [info] LORA: Changing state from 0 to 0
[2018-05-12T08:51:05.610Z] [info] MAIN: Leaving state 0
[2018-05-12T08:51:05.612Z] [info] MAIN: Entering state 1
[2018-05-12T08:51:05.620Z] [info] FLASH: Magic bytes found: wifi config present
[2018-05-12T08:51:05.626Z] [info] FLASH: Magic bytes found: activation data present
[2018-05-12T08:51:05.638Z] [info] FLASH: Magic bytes found: FOTA data present
[2018-05-12T08:51:05.639Z] [info] FLASH: Loading Firmware Data
[2018-05-12T08:51:05.645Z] [info] CNFG: (Firmware HASH (sha256)) 14 3E 99 52 CC 5B 11 E1 A2 21 8A 04 06 AA 73 1D 1D D8 25 CE FF BD C3 28 34 1E FB 0B B2 AC 9B 56
[2018-05-12T08:51:05.648Z] [info] FLASH: Loading WiFi Data
[2018-05-12T08:51:05.656Z] [info] CNFG: WiFi SSID:      <redacted>
[2018-05-12T08:51:05.659Z] [info] CNFG: WiFi key:       ***
[2018-05-12T08:51:05.660Z] [info] CNFG: WiFi conn_type: 1
[2018-05-12T08:51:05.661Z] [info] CNFG: WiFi sec_type:  4
[2018-05-12T08:51:05.662Z] [info] FLASH: Loading Activation Data
[2018-05-12T08:51:05.664Z] [info] CNFG: Gateway ID:         arjanvanb-gw-1
[2018-05-12T08:51:05.667Z] [info] CNFG: Gateway Key:        ***
[2018-05-12T08:51:05.672Z] [info] CNFG: Account Server URL: https://account.thethingsnetwork.org
[2018-05-12T08:51:05.674Z] [info] CNFG: Locked:             true
[2018-05-12T08:51:05.677Z] [info] CNFG: Locked first time:  false
[2018-05-12T08:51:05.680Z] [info] MAIN: Leaving state 1
[2018-05-12T08:51:05.682Z] [info] MAIN: Entering state 2
[2018-05-12T08:51:05.684Z] [info] INET: State change to 0
[2018-05-12T08:51:05.985Z] [info] LORA: Initialisation complete
[2018-05-12T08:51:05.988Z] [info] LORA: Changing state from 0 to 1
[2018-05-12T08:51:06.240Z] [info] WIFI: Entering state 1
[2018-05-12T08:51:06.242Z] [info] ETH: IP Address: 0.0.0.0
[2018-05-12T08:51:06.246Z] [info] WIFI: Entering state 4
[2018-05-12T08:51:06.250Z] [info] WIFI: Entering SCAN state 1
[2018-05-12T08:51:07.335Z] [info] SNTP: State change from 0 to 1
[2018-05-12T08:51:07.385Z] [info] INET: Gateway has Ethernet
[2018-05-12T08:51:07.387Z] [info] INET: State change to 2
[2018-05-12T08:51:07.393Z] [warn] INET: Connected to a network, waiting for DHCP lease, checking validity with ping
[2018-05-12T08:51:08.433Z] [info] SNTP: State change from 1 to 2
[2018-05-12T08:51:08.536Z] [info] Scan is completed successfully
[2018-05-12T08:51:08.539Z] [info] WIFI: Entering SCAN state 2
[2018-05-12T08:51:08.546Z] [info] WIFI: Entering SCAN state 5
[2018-05-12T08:51:08.549Z] [info] WIFI: Entering SCAN state 0
[2018-05-12T08:51:08.552Z] [info] WIFI: Entering state 2
[2018-05-12T08:51:08.554Z] [info] WIFI: Disabling modules
[2018-05-12T08:51:08.559Z] [info] Head magic match void: trying to free an already freed block, ignore
[2018-05-12T08:51:08.562Z] [info] SNTP: State change from 2 to 1
[2018-05-12T08:51:08.570Z] [info] WIFI: Entering state 3
[2018-05-12T08:51:09.431Z] [info] WIFI: Enabling modules for client
[... 570 lines cut ...]
[2018-05-12T15:11:28.515Z] [info] MQTT: Sending UPLINK OK
[2018-05-12T15:11:28.519Z] [info] MQTT: Sending status packet
[2018-05-12T15:11:28.523Z] [warn] MQTT: Report reboot error: 0103
[2018-05-12T15:11:28.531Z] [warn] MQTT: Report config error: 0600000020
[2018-05-12T15:11:28.547Z] [info] MQTT: Sending status succeeded: 0
[2018-05-12T15:11:32.770Z] [info] SNTP: State change from 5 to 6
[2018-05-12T15:11:47.049Z] [info] SNTP: State change from 6 to 1
[2018-05-12T15:11:48.147Z] [info] SNTP: State change from 1 to 2
[... cut ...]
[2018-05-12T15:11:51.453Z] [info] SNTP: State change from 6 to 7
[2018-05-12T15:11:58.016Z] [info] LORA: Accepted packet
[2018-05-12T15:11:58.054Z] [info] MQTT: Sending UPLINK OK
[2018-05-12T15:11:58.502Z] [info] MQTT: Sending status packet
[2018-05-12T15:11:59.541Z] [info] MQTT: Sending status succeeded: 1
[2018-05-12T15:12:06.327Z] [info] LORA: Accepted packet
[2018-05-12T15:12:06.356Z] [info] MQTT: Sending UPLINK OK
[2018-05-12T15:12:22.754Z] [info] LORA: Accepted packet
[2018-05-12T15:12:22.782Z] [info] MQTT: Sending UPLINK OK
```
</details>

avbentem, 2018-06-03:
If my notes don't fail me, then since my May 12th comment I've pressed the Mode button to resolve this problem:

- May 15th, 7:25 Amsterdam time
- May 22nd, 6:45
- May 22nd, 18:25
- May 27th, 22:20
- June 3rd, 9:08, 9:13, 9:15

So today **it was again running fine until it did its daily reboot for a firmware upgrade check, after which the LoRa module just failed to receive any LoRa packets**. Today I had to press the Mode button 3 times to get things working again. Regardless wether or not receiving LoRa packets, only the main (power?) LED is illuminated on the card, while during the restart the LED marked "1" (in the row of 5) was on one or two times shortly too.

(Using ethernet though WiFi is configured too. And still running 1.0.3 from an SD card, commit: 2c56ce09, timestamp: 1521721252, Build time: Mar 22 2018 12:21:52, as 1.0.4 is giving me a reboot loop. As an aside, it's not clear to me if 1.0.3 or 1.0.4 is the current stable release if one would not use an SD card.)

KrishnaIyer, 2018-07-25:
This is related to #47 as the gateway controller was unable to maintain UART connection with the LoRa module and is also addressed in `1.0.5`. 
Once again, please feel free to re-open it if the issue persists in `1.0.5`.

closed 2018-07-25 by KrishnaIyer

avbentem, 2018-08-05:
I've indeed not run into this issue since I updated to 1.0.5 on July 17th.

(Last occurrence on Monday July 16th, using 1.0.3.)