TheThingsProducts/gateway #33: Gateway losing client connection to AP, defaulting back to Things-Gateway-xxx

aderusha, 2018-04-07:
I've deployed a new gateway and configured the device for my local WiFi.  This works as expected for a while, then drops connection to WiFi until manually power cycled.  I've connected a UART to collect debug logs which show the device assigning itself IP 192.168.84.1 and starting the AP when this happens which is the default behavior when it cannot connect to the assigned AP but otherwise working as expected. 

For testing purposes I enabled beta firmware updates, which ran, and which haven't changed the resulting behavior.  When I connect to the Things-Gateway-xxx and watch the logs, I can see it directly stating that it's keeping the AP configured due to a client being connected (`INET: Not trying to connect to WiFi router again because client is connected (after 120 seconds)`).  When I disconnect my client from that AP, that message does not appear again, but it still drops off of the assigned SSID and defaults back to acting as an AP (again, without mentioning a client being connected).

I have recorded logs from the original firmware (factory? - I don't see where to collect the firmware details), then showing the upgrade to beta, showing one instance of a client being connected to the AP, and a whole lot of failing to connect to the assigned AP.  When you see "10.0.100.20", that's the gateway connecting successfully to my internal network.

## To recap:

- Environment has 3 APs, good signal strength and channel separation
- TTN Gateway will connect to the assigned SSID immediately without problem
- Some time later, it will drop that connection and go into AP mode with no obvious reason given as to why
- No clients are connected to the Things-Gateway-xxx AP

I am working on a city-wide deployment of these devices.  This is the first device to be tested and right now we are at a hard stop with this deployment until this issue can be resolved.

aderusha, 2018-04-07:
Link to logs: https://hastebin.com/yimidajica

Sebas-, 2018-04-10:
+1 for priority on the WIFI instability.

My gateway is located on a place without LAN, so the WIFI feature is really good.. except that it drops the connection a lot, and then reverts to AP mode. My impression is that it simply does not try to reconnect (or it gives up way to easy). 

Currently this problem is solved by the instability of the gateway itself, because when it "automatically" reboots it reconnects to the WIFI (as an additional workaround I also added a WIFI connected 220v relais so I don't have to get to the AP every single time it stays down longer than expected, the WIFI connection in the relais (ESP8266) is very stable :) ).

PR #38 (2018-04-21): Fix/keep existing wifi config
Copy of commit message:
```
The previous WiFi patch that restricted AP mode to only when there is no WiFi config causes issues with gateways with exisiting WiFi config.
Though the two are not directly related, the patch seems to have uncovered an existing problem.
In the current code, the WiFi ping timeout is 8s (4x2 tries). But practically, it takes 10-15s to re-connect. So, after 8s, we go to WAIT_FOR_NETWORK state where the WiFi module is restarted.
This causes a loop since everytime the pings timeout at 8s and restart the WiFi module.
By increasing the ping ti [... cut]
closed 2018-04-21 by KrishnaIyer