TheThingsProducts/gateway #35: US_915 Gateway shipped as EU_868

KrishnaIyer, 2018-04-12:
This is a **bug report**

### What do you want to do? What do you expect?
Connect my gateway to TTN. Gateway is packaged as EU_868.

### What steps did you take? (Give as many details as possible so that we can reproduce it)
* Connect Gateway to power supply.
* Activate using the Activation procedure(frequency plan EU_868)

### What went wrong? What is missing?
* LED 3 fast blink. Not activated. 

### What firmware is running on your gateway?

### What do the logs show?
*  LORA: ERROR: wrong frequency configuration, fallback on default
[SerialLogs.txt](https://github.com/TheThingsProducts/gateway/files/1902963/SerialLogs.txt)

### Any Screenshots or Photos?
[image]

### Do you think you can try to fix this yourself and submit a pull request?

The gateway should work fine if configured to US_915, but this is a hardware labeling mismatch and not a software issue.

KrishnaIyer, 2018-04-12:
ping @wienke

rish-c, 2018-04-13:
Fixed

closed 2018-04-13 by rish-c