TheThingsProducts/gateway #47: Boot stops in non-working state when failing to get configuration (LoRa module and bridging not enabled)

avbentem, 2018-06-09:
This is a **bug report**.

### What do you want to do? What do you expect?

I want/expect my gateway to run all day without any human intervention.

### What steps did you take? (Give as many details as possible so that we can reproduce it)

Gateway was running just fine...

### What went wrong? What is missing?

...until it suddenly rebooted for no apparent reason, and during the boot could not get a configuration but reported `CONF: ERROR REQUEST`, `CNFG: Downloading gateway configuration failed` and `CNFG: Communication ERROR`. (For this, network problems seem unlikely as it was reporting through MQTT just fine before the reboot? Maybe the remote server was down?) 

After that it did not properly finish its boot. Like: it never even tried to get the configuration again, nor continued with its existing configuration. So, it was just not operational at all and did not bother to resolve that.

### What firmware is running on your gateway?

Loaded from an SD card:

AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1521721252
Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
Build time: Mar 22 2018 12:21:52

Connected through Ziggo ethernet, while XS4ALL WiFi is configured too.

### What do the logs show?

All timestamps in UTC (as added by my own monitor).

First, I've no clue why it even restarted, but that happens more often so is likely unrelated to this issue:

```text
[2018-06-07T04:39:16.060Z] LORA: Accepted packet
[2018-06-07T04:39:16.090Z] MQTT: Sending UPLINK OK
[2018-06-07T04:39:16.450Z] MQTT: Received DOWNLINK
[2018-06-07T04:39:25.346Z] SNTP: State change from 0 to 0
[2018-06-07T04:39:25.350Z] SNTP: State change from 0 to 0
[2018-06-07T04:39:25.352Z] 
[2018-06-07T04:39:25.353Z] 
[2018-06-07T04:39:25.354Z] 
[2018-06-07T04:39:25.356Z] **************************
[2018-06-07T04:39:25.358Z] *   The Things Network   *
[2018-06-07T04:39:25.361Z] *      G A T E W A Y     *
[2018-06-07T04:39:25.363Z] **************************
[2018-06-07T04:39:25.369Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.3, commit: 2c56ce09, timestamp: 1521721252
[2018-06-07T04:39:25.405Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-06-07T04:39:25.407Z] Build time: Mar 22 2018 12:21:52
[2018-06-07T04:39:25.408Z] Reboot reason: 0x10
[2018-06-07T04:39:25.419Z] BOOT: (persisted info) 6F 72 72 65 01 10 83 2C E7 16 09 49 92 3A 51 B7 
```

A bit later, it successfully gets a network connection:

```text
[2018-06-07T04:39:30.402Z] CB: ETH: IP Address: 192.168.0.12 
[2018-06-07T04:39:32.373Z] INET: State change to 3
[2018-06-07T04:39:32.376Z] INET: Ping probe
[2018-06-07T04:39:32.386Z] INET: Ping response from PIC32INT, set as default
[2018-06-07T04:39:32.389Z] INET: State change to 4
[2018-06-07T04:39:32.398Z] WIFI: IP Address: 192.168.178.53 
[2018-06-07T04:39:32.961Z] LORA: Wait init complete, check for communication on 115200 baud.
[2018-06-07T04:39:32.964Z] LORA: Changing state from 1 to 2
[2018-06-07T04:39:32.970Z] LORA: Communication on 115200 OK, waiting for application.
[2018-06-07T04:39:32.973Z] LORA: Changing state from 2 to 4
[2018-06-07T04:39:33.892Z] SNTP: State change from 2 to 3
[2018-06-07T04:39:34.990Z] SNTP: State change from 3 to 4
[2018-06-07T04:39:35.329Z] MON: SYS Stack size: 3965
[2018-06-07T04:39:35.332Z] MON: heap usage: 148KB (156KB), free: 191KB
[2018-06-07T04:39:36.093Z] SNTP: State change from 4 to 5
[... 674 lines cut ...]
[2018-06-07T21:16:39.401Z] 
[2018-06-07T21:16:39.405Z] *************************
[2018-06-07T21:16:39.407Z] MAIN: Gateway bridging
[2018-06-07T21:16:39.412Z] *************************
[2018-06-07T21:16:39.414Z] 
[2018-06-07T21:16:39.424Z] MQTT: Sending status packet
[2018-06-07T21:16:39.426Z] MQTT: Report reboot error: 0140
[2018-06-07T21:16:39.433Z] MQTT: Sending status succeeded: 0
[2018-06-07T21:16:43.815Z] LORA: Packet dropped! Bad CRC
[2018-06-07T21:16:43.883Z] LORA: Accepted packet
[2018-06-07T21:16:43.912Z] MQTT: Sending UPLINK OK
[2018-06-07T21:16:48.082Z] MON: SYS Stack size: 2870
[2018-06-07T21:16:48.086Z] MON: heap usage: 263KB (266KB), free: 76KB
[2018-06-07T21:16:50.543Z] LORA: Accepted packet
[2018-06-07T21:16:50.571Z] MQTT: Sending UPLINK OK
```
</details>

### Any Screenshots or Photos?

Not applicable.

### Do you think you can try to fix this yourself and submit a pull request?

No.

KrishnaIyer, 2018-07-25:
This issue was the focal point of our `1.0.5` fix. I'm closing this since we have enough positive feedback. Please feel free to re-open it if you experience it in the latest stable build.

closed 2018-07-25 by KrishnaIyer

avbentem, 2018-08-05:
This is still happening in 1.0.5.

I've upgraded from 1.0.3 to 1.0.5 on Tuesday July 17th, around 9 PM Amsterdam time. This is using the factory bootloader. Since the upgrade my gateway has run into `*** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking` 225 times; see #52. In 3 of those 225 occurences it failed to recover after it rebooted: it got an ethernet connection, and showed `HTTP: Starting connection` but never showed anything about the network after that (like it did NOT show `Gateway bridging`), though it was receiving LoRa packets. A manual reboot using the pink Mode button fixed this. 

Apart from the following 3 occasions, the gateway has rebooted many, many more times, like after an MQTT error and also for the 222 other occurences of the above assert. But for those reboots, it recovered just fine.

<details>
<summary>Saturday July 21st, no gateway bridging after reboot, until manual reboot, in 1.0.5</summary>

```txt
[2018-07-21T05:07:09.551Z] MQTT: Sending status packet
[2018-07-21T05:07:09.585Z] MQTT: Sending status succeeded: 671
[2018-07-21T05:07:13.413Z] MON: SYS Stack size: 2837
[2018-07-21T05:07:13.416Z] MON: TCPIP Stack size: 3771
[2018-07-21T05:07:13.418Z] MON: APP Stack size: 3271
[2018-07-21T05:07:13.420Z] MON: LoRa Stack size: 3857
[2018-07-21T05:07:13.424Z] MON: heap usage: 282KB (283KB), free: 56KB
[2018-07-21T05:07:17.288Z] LORA: Kic
[2018-07-21T05:07:17.294Z] *** assert ../src/app_lora.c:842:w == size:LORA Uart write should be blocking
[2018-07-21T05:07:17.297Z] 
[2018-07-21T05:07:19.324Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.328Z] SNTP: State change from 0 to 0
[2018-07-21T05:07:19.331Z] 
[2018-07-21T05:07:19.332Z] 
[2018-07-21T05:07:19.333Z] 
[2018-07-21T05:07:19.334Z] **************************
[2018-07-21T05:07:19.337Z] *   The Things Network   *
[2018-07-21T05:07:19.339Z] *      G A T E W A Y     *
[2018-07-21T05:07:19.342Z] **************************
[2018-07-21T05:07:19.346Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-07-21T05:07:19.367Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-07-21T05:07:19.369Z] Build time: Jul 17 2018 08:12:51
[2018-07-21T05:07:19.382Z] Reboot reason: 0x40
[2018-07-21T05:07:19.391Z] BOOT: (persisted info) 6F 72 72 65 01 40 AB 2C E7 16 05 C9 B2 3A 50 97 
[2018-07-21T05:07:19.393Z] 
[2018-07-21T05:07:19.394Z] 
[2018-07-21T05:07:19.395Z] 
[2018-07-21T05:07:19.396Z] 
[2018-07-21T05:07:19.397Z] WIFI: Entering state 0
[2018-07-21T05:07:19.399Z] WIFI: Entering SCAN state 0
[2018-07-21T05:07:19.400Z] 
[2018-07-21T05:07:19.401Z] MAIN: Initialisation complete
[2018-07-21T05:07:19.415Z] LORA: Changing state from 0 to 0
[2018-07-21T05:07:19.570Z] 
[2018-07-21T05:07:19.573Z] MAIN: Leaving state 0
[2018-07-21T05:07:19.575Z] MAIN: Entering state 1
[2018-07-21T05:07:19.582Z] FLASH: Magic bytes found: wifi config present
[2018-07-21T05:07:19.589Z] FLASH: Magic bytes found: activation data present
[2018-07-21T05:07:19.593Z] FLASH: Magic bytes found: FOTA data present
[2018-07-21T05:07:19.596Z] FLASH: Loading Firmware Data
[2018-07-21T05:07:19.607Z] CNFG: (Firmware HASH (sha256)) 93 F4 C1 0E 77 81 3F 60 0D A8 A5 97 57 7C DD 9C 95 64 1F DF F4 8D 44 3F 92 9A 20 A9 2D B4 B9 F5 
[2018-07-21T05:07:19.610Z] FLASH: Loading WiFi Data
[2018-07-21T05:07:19.613Z] CNFG: WiFi SSID:      <redacted>
[2018-07-21T05:07:19.616Z] CNFG: WiFi key:       ***
[2018-07-21T05:07:19.618Z] CNFG: WiFi conn_type: 1
[2018-07-21T05:07:19.621Z] CNFG: WiFi sec_type:  4
[2018-07-21T05:07:19.623Z] FLASH: Loading Activation Data
[2018-07-21T05:07:19.626Z] CNFG: Gateway ID:         arjanvanb-gw-1
[2018-07-21T05:07:19.629Z] CNFG: Gateway Key:        ***
[2018-07-21T05:07:19.635Z] CNFG: Account Server URL: https://account.thethingsnetwork.org
[2018-07-21T05:07:19.638Z] CNFG: Locked:             true
[2018-07-21T05:07:19.641Z] CNFG: Locked first time:  false
[2018-07-21T05:07:19.643Z] 
[2018-07-21T05:07:19.645Z] MAIN: Leaving state 1
[2018-07-21T05:07:19.647Z] MAIN: Entering state 2
[2018-07-21T05:07:19.650Z] INET: State change to 0
[2018-07-21T05:07:19.947Z] LORA: Initialisation complete
[2018-07-21T05:07:19.950Z] LORA: Changing state from 0 to 1
[2018-07-21T05:07:20.202Z] WIFI: Entering state 1
[2018-07-21T05:07:20.205Z] ETH: IP Address: 0.0.0.0 
[... 394 lines cut ...]
[2018-08-02T00:38:59.802Z] LGMD:LORA: Accepted packet
...
...
[2018-08-02T04:32:05.113Z] LGMD:Rejected packet (0x11)
[2018-08-02T04:32:05.116Z] 
[2018-08-02T04:32:05.956Z] SNTP: State change from 1 to 2
[2018-08-02T04:32:06.897Z] BTN: change to state: 1
[2018-08-02T04:32:06.998Z] BTN: change to sta
[2018-08-02T04:32:07.001Z] *** Application reboot
[2018-08-02T04:32:07.008Z] 
[2018-08-02T04:32:09.030Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.033Z] SNTP: State change from 0 to 0
[2018-08-02T04:32:09.035Z] 
[2018-08-02T04:32:09.037Z] 
[2018-08-02T04:32:09.038Z] 
[2018-08-02T04:32:09.040Z] **************************
[2018-08-02T04:32:09.042Z] *   The Things Network   *
[2018-08-02T04:32:09.045Z] *      G A T E W A Y     *
[2018-08-02T04:32:09.047Z] **************************
[2018-08-02T04:32:09.052Z] Firmware name: AmazingAckermann, type: 0, version: 1.0.5, commit: fa89b993, timestamp: 1531815112
[2018-08-02T04:32:09.074Z] Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
[2018-08-02T04:32:09.077Z] Build time: Jul 17 2018 08:12:51
[2018-08-02T04:32:09.091Z] Reboot reason: 0x40
```
</details>

danielkucera, 2022-07-26:
I think this is more related to #71