TheThingsProducts/gateway #49: Getaways not being able to activate and rebooting itself.

nicoyuste, 2018-07-02:
This is a bug report:

### What do you want to do? What do you expect?
I am trying to connect my new Getaways to The Things Network but it does not connect at all.

### What steps did you take? (Give as many details as possible so that we can reproduce it)
I opened this website: https://activate.thethingsnetwork.org/ and followed all the steps. Every goes fine while creating the ID for the Getaway and setting up the Wifi details or Ethernet (I tried both). Once everything is correct and I send to the Getaway the information it tries to connect to the things network but it never finishes.

### What went wrong? What is missing?
It restarts itself all the time without being able to activate, here you have a video of the LEDs on my getaway: https://youtu.be/Yu-ek70Iuyc

### What firmware is running on your gateway?
How can I know that?

### What do the logs show?
How can I get the logs?

### Any Screenshots or Photos?
Yes, this video: https://youtu.be/Yu-ek70Iuyc

### Do you think you can try to fix this yourself and submit a pull request?
No.

nicoyuste, 2018-07-02:
I was finally able to connect it to the network. There was a hardware problem (I read online) and I had to firmly press the Microchip a little bit. We can resolve this issue now.

KrishnaIyer, 2018-07-02:
@nicoyuste : thanks.

closed 2018-07-02 by KrishnaIyer