TheThingsProducts/gateway #51: Error: Request has been terminated.

victorpinon, 2018-07-09:
This is a **bug report.**

### What do you want to do? What do you expect?
I want to activate my gateway (the things gateway backer edition).

### What steps did you take? (Give as many details as possible so that we can reproduce it)
I followed the steps on https://activate.thethingsnetwork.org/ until I got to step 3 ans the error appeared.

### What went wrong? What is missing?
This error showed:
**Request has been terminated.
Possible causes 
the Network is offline, Origin is not allowed by Access-Control-Allow-Origin, the page is being unloaded, etc.**

### What firmware is running on your gateway?
 v.1.0.2-475aef56

### Any Screenshots or Photos?
[image]
[image]

### Do you think you can try to fix this yourself and submit a pull request?
Unfortunately no.

KrishnaIyer, 2018-07-09:
@victorpinon Are you using only ethernet as the backhaul?

victorpinon, 2018-07-09:
@KrishnaIyer  Yes. But tried with Wifi and had the same error.

KrishnaIyer, 2018-07-10:
Hmmm. Do you by any chance have serial logs?

victorpinon, 2018-07-10:
@KrishnaIyer I will get the logs later today or tomorrow.
I also made this video https://youtu.be/dSRto_uJN1M

victorpinon, 2018-07-12:
@KrishnaIyer How do I get serial logs using Windows?

KrishnaIyer, 2018-07-12:
Oh windows?
Well if you're using a USB<-->UART, then the easiest would be to use a hyperterminal (Ex: Putty, if it still exists) and connect to the COM port that the USB is read out of. 
Baudrate: `115200`

victorpinon, 2018-07-13:
SNTP: State change from 0 to 0
SNTP: State change from 0 to 0

**************************
*   The Things Network   *
*      G A T E W A Y     *
**************************
Firmware name: AmazingAckermann, type: 0, version: 1.0.2, commit: 475aef56, timestamp: 1521106500
Bootloader revision: 1, commit: 7167873a, timestamp: 1496411298
Build time: Mar 15 2018 09:35:17
Reboot reason: 0x40
BOOT: (persisted info) 6F 72 72 65 01 03 F1 C4 48 35 91 94 6E 98 4E F8

WIFI: Entering state 0
WIFI: Entering SCAN state 0

MAIN: Initialisation complete
LORA: Changing state from 0 to 0

MAIN: Leaving state 0
MAIN: Entering state 1
FLASH: Magic bytes not found: no stored wifi config present
FLASH: Magic bytes not found: no stored activation data present
FLASH: Magic bytes not found: no stored FOTA data present
FLASH: Loading Firmware Data
CNFG: (Firmware HASH (sha256)) 87 93 F2 0A DA 7C FB 56 3C EB E0 29 21 C5 BB 76 F3 9C A6 A4 A9 2C CE 01 EC A8 29 5B 59 48 9C 20

MAIN: Leaving state 1
MAIN: Entering state 2
INET: State change to 0
INET: No Ethernet and no WiFi config
INET: State change to 1
LORA: Initialisation complete
LORA: Changing state from 0 to 1
WIFI: Entering state 1
ETH: IP Address: 0.0.0.0
WIFI: Entering state 4
WIFI: Entering SCAN state 1
Scan is completed successfully
WIFI: Entering SCAN state 2
WIFI: Entering SCAN state 5
WIFI: Entering SCAN state 0
WIFI: Entering state 2
WIFI: Disabling modules
Head magic match void: trying to free an already freed block, ignore
WIFI: Entering state 3
SNTP: State change from 0 to 1
INET: State change to 0
INET: Gateway has Ethernet
INET: State change to 2
INET: Connected to a network, waiting for DHCP lease, checking validity with ping
WIFI: Entering state 4
WIFI: Entering SCAN state 1
Scan is completed successfully
WIFI: Entering SCAN state 2
WIFI: Entering SCAN state 3
Head magic match void: trying to free an already freed block, ignore
SNTP: State change from 1 to 1
WIFI: Entering SCAN state 4
SNTP: State change from 1 to 2
WIFI: Entering SCAN state 5
WIFI: Entering SCAN state 0
WIFI: Entering state 5
WIFI: Enabling modules for server
WIFI: Entering state 6

>WIFI: IP Address: 192.168.84.1
ETH: IP Address: 192.168.1.133
CB: INET: State change to 3
INET: Ping probe
INET: Error sending probe on WiFi
INET: Ping response from PIC32INT, set as default
INET: State change to 4
LORA: Wait init complete, waiting for application.
LORA: Changing state from 1 to 2
SNTP: State change from 2 to 3
SNTP: State change from 3 to 4
MON: SYS Stack size: 3959
MON: heap usage: 147KB (160KB), free: 192KB
SNTP: State change from 4 to 5
SNTP: State change from 5 to 6
SNTP: State change from 6 to 7
INET: State change to 5

MAIN: Leaving state 2
MAIN: Entering state 3

CNFG: Load online user config state change to 0
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB

KrishnaIyer, 2018-07-14:
Well the gateway seems to be downloading a new Firmware. Best thing to do would be to allow the process to complete.

victorpinon, 2018-07-15:
It has been turned on for 12 hours and it seems that it is doing the same all the time.
It also shows this sometimes:

MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB
SNTP: State change from 7 to 1
SNTP: State change from 1 to 2
SNTP: State change from 2 to 3
SNTP: State change from 3 to 4
SNTP: State change from 4 to 5
SNTP: State change from 5 to 6
SNTP: State change from 6 to 7
MON: SYS Stack size: 3959
MON: heap usage: 148KB (160KB), free: 192KB

KrishnaIyer, 2018-07-15:
The `MON:` logs are from the system monitor. Seems like there's some trouble connecting to the internet and it's not able to download the newer firmware. Can you attach a full log file from start-up?

victorpinon, 2018-07-15:
[Logs.txt](https://github.com/TheThingsProducts/gateway/files/2195600/Logs.txt)
This is all it shows, I can restart to get the first ones again if you want.

victorpinon, 2018-07-15:
[Logs2.txt](https://github.com/TheThingsProducts/gateway/files/2196062/Logs2.txt)
Here it shows the first ones but with less time running

KrishnaIyer, 2018-07-16:
Both logs seems to show that the Gateway is not able to connect to the internet due to some reason. This is an issue that we've seen in some gateways and our `beta-1.0.4` has a fix that should stabilize the network.

Assuming that you've registered the gateway and enabled the Beta updates, the problem now is that since there is no stable connection, the newer version is not being downloaded.

If you have an SD Card, could to try to manually install the newer firmware using the procedure below:
`https://github.com/TheThingsProducts/gateway/tree/develop/firmware#installation`

In that case, please use the `beta` firmware from the links below:
`https://github.com/TheThingsProducts/gateway/tree/develop/firmware#firmware-releases`

victorpinon, 2018-07-16:
Installed the firmware. Still the same error.
[image]
[Logs3.txt](https://github.com/TheThingsProducts/gateway/files/2198416/Logs3.txt)

rish-c, 2018-07-23:
Hi Víctor, can you please drop me an email? (info@thethingsproducts.com)