TheThingsProducts/gateway #54: Add ISRG Root X1 Certificate to trusted roots

htdvisser, 2018-10-05:
This is a **feature request**

### What do you want to do? What do you expect?

I think we should prepare for the widespread adoption of Let's Encrypt TLS certificates, and should be prepared for when more servers choose to serve Let's Encrypt certificates with the cert chain that includes `ISRG Root X1` instead of the IdenTrust `DST Root CA X3`. 

### What went wrong? What is missing?

The `ISRG Root X1` root is [not trusted](https://github.com/TheThingsProducts/gateway/blob/master/firmware/src/system_config/TTN_Gateway_v1/framework/net/pres/net_pres_cert_store.c) by the Gateway.

### What firmware is running on your gateway?

`1.0.5`

### What do the logs show?

Not applicable

### Any Screenshots or Photos?

Not applicable

### Do you think you can try to fix this yourself and submit a pull request?

Maybe, but that could take quite some time. I guess it's better if @KrishnaIyer does it (maybe with help of @martinichka).

avbentem, 2020-07-27:
It seems this was solved Apr 18, 2019 and releases in 1.0.6?

- https://github.com/TheThingsProducts/gateway/releases/tag/v1.0.6

- Hylke [on the forum](https://www.thethingsnetwork.org/forum/t/automatic-gateway-update/32990/14):

    > The `1.0.6` patch with the ISRG Root X1 has been on the `beta` channel since May 2019, and on the `stable` channel since June 2019.

- Commit history [for `net_pres_cert_store.c`](https://github.com/TheThingsProducts/gateway/commits/master/firmware/src/system_config/TTN_Gateway_v1/framework/net/pres/net_pres_cert_store.c) (stable channel)

htdvisser, 2020-07-27:
Yeah, thanks, this one can be closed.

closed 2020-07-27 by htdvisser