TheThingsProducts/gateway #55: Web UI rejects valid Gateway ID

htdvisser, 2018-11-07:
This is a **bug report** initially reported as https://github.com/TheThingsNetwork/docs/issues/276

### What do you want to do? What do you expect?

The Gateway's web UI should accept all valid gateway IDs

### What steps did you take? (Give as many details as possible so that we can reproduce it)

- Activate the gateway as `gw0`
- Try changing the WiFi network in the Gateway's web page

### What went wrong? What is missing?

- The form validation fails on ID `gw0`

### What firmware is running on your gateway?

unknown

### What do the logs show?

unknown

### Any Screenshots or Photos?

unknown

### Do you think you can try to fix this yourself and submit a pull request?

@stapelberg please reply

stapelberg, 2018-11-07:
Thanks for routing this the right direction. I didn’t realize the firmware for the gateway was open and available. I’ll try to send a PR for this soon.

stapelberg, 2018-11-10:
Change of plans: I won’t have an opportunity to look at this for the foreseeable future. Sorry!

htdvisser, 2018-11-12:
No problem. After some more investigation we decided that this isn't really a big issue, and it has an easy workaround, so we'll leave this here to pick up when someone has time for it.