TheThingsProducts/gateway #56: Serial Log avilable on SSH port, feature request

trlafleur, 2018-11-13:
This is a feature request

it would be very convenient is the serial logs could be available also on an SSH port from the gateway..

This would allow capturing the logs without the need for a physical connection to the device...

thank you...

avbentem, 2019-01-29:
The TTN gateway is not some Linux device. It does not support SSH at all. The web interface might be made to expose the logs though? (And when doing that, more details about the LoRa packets would be great too.)

KrishnaIyer, 2019-02-04:
As @avbentem mentioned there's no SSH support since it's baremetal and not a linux based device.

The Web Interface is exposed only when the gateway switches to AP mode. We can hack it to be active during traffic mode but this adds additional complexity for the multiplexing.

We're not planning this right now but we are very open to community contributions.