TheThingsProducts/gateway #57: Support for static ip setup

lanmarc77, 2019-01-10:
Please allow for setting up a static ip address, gw and dns server.
This could be done via a special web server url e.g. /staticip or even via the main page.

johanstokking, 2019-01-10:
We considered this, but we decided against this for the reason of simplicity for the users (and obviously for us too) and we did not see a use case for this. The web interface is exposed in AP mode, and then the IP address is static, otherwise there's no reason to connect to the gateway.

lanmarc77, 2019-01-10:
I do have a use case. The gateway will be placed in an existing DMZ. To minimize the attack vectors no DHCP service is available there. I would not need the static IP to access the gateway but to be able to assign one in the first place. From my experience this is a very common setup in existing it infrastructures.
For security reasons it would be helpful to also be able to disable the web interface as well. Knowing there is no way to access it again a reset of the ttn gateway would be needed.
The normal user would not have it harder when an additional url is used that only experts know.
I do not know who WE is but would really like you to reconsider as I think it will only support the movement to get more gateways up. I assume most of the gateways will be brouht up by experts in existing infrastructures anyways. The more options they have the better. But for the normal user always start in "normal" mode.

johanstokking, 2019-01-10:
Do you mean DMZ in the way that the gateway is placed in a zone which can access the public internet, i.e. a TTN network, while other nodes cannot?

I see the use case, but the gateway was not designed with these network setups in mind.

We are, however, open to community contributions. Hence, I'm reopening the issue and welcome contributions to the code base from the community. It shouldn't be rocket science, although we'd need a user interface for it.

lanmarc77, 2019-02-18:
I do like the simplistic interface and would not change anything. How about an configuration file on the SD card, not only for static IP adresses but also for disabling the wirless interface and webserver and whatever else the "not normal" people want? I already looked at the code and I try to get students at my university motivated to have there thesis on this topic.