TheThingsProducts/gateway #59: Problems building and uploading firmware using MPLAB

frankminfarm, 2019-03-22:
This is a **question**

### What do you want to do? What do you expect?

I am trying to build and upload the TTN-Gateway firmware code using MPLAB X IDE v3.45. While the project seems to build fine - see log below - and the firmware is uploaded using my PICkit3 (Chinese copy), none of the TTN-Gateway's LEDs turn on and the board is unresponsive.

### What steps did you take? (Give as many details as possible so that we can reproduce it)

1. I followed the steps documented [here](https://github.com/TheThingsProducts/gateway/tree/develop/firmware).

2. However, running compile.sh failed with the following output:

`make: *** [.build-conf] Error 2`

3. I then opened up the project in MPLAD IDE and ran it from there with the results outlined above - the project seems to build fine and the hex file is uploaded to the TTN-G.

### What went wrong? What is missing?

None of the TTN-Gateway's LEDs turn on and the board is unresponsive once the built firmware is uploaded.

### What firmware is running on your gateway?

I believe that it's v1.0.5 - at least this is the output of running the generate_version_header.sh command - is this correct?

I have been able to upload the latest stable firmware .hex file using MPLAB IPE and this works fine - i.e. the TTN-G is operational.

### What do the logs show?

Here is the output from MPLAB IDE - note that I have shortened this due to the post size limit restriction:

````
make -f nbproject/Makefile-TTN_Gateway_v1.mk SUBPROJECTS= .build-conf
make[1]: Entering directory '/Users/user/Documents/company/projects/ttn/gateway/gateway/firmware/TTN_Gateway.X'
make  -f nbproject/Makefile-TTN_Gateway_v1.mk dist/TTN_Gateway_v1/debug/TTN_Gateway.X.debug.elf
make[2]: Entering directory '/Users/user/Documents/company/projects/ttn/gateway/gateway/firmware/TTN_Gateway.X'
"/Applications/microchip/xc32/v1.42/bin/xc32-gcc" -g -D__DEBUG -D__MPLAB_DEBUGGER_PK3=1 -fframe-base-loclist -x c -c -mprocessor=32MZ2048EFM144 -ffunction-sections -O1 -D_FULL_SIGNAL_IMPLEMENTATION -I../src -I../vendor/harmony/framework -I../vendor/jsmn -I../vendor/json-builder -I../vendor/GenericRecorderLibSrc/Include -I../vendor/ttn-gateway-connector/src -I../vendor/protobuf-c -I../vendor/ttn-gateway-connector/src/github.com/TheThingsNetwork -I../vendor/ttn-gateway-connector/src/github.com/gogo/protobuf/protobuf -I../vendor/paho.mqtt.embedded-c/MQTTPacket/src -I../vendor/paho.mqtt.embedded-c/MQTTClient-C/src -I../vendor/mbedtls/include -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1 -I../src/TTN_Gateway_v1 -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1/framework -I../vendor/harmony/framework/driver/wifi/mrf24wn/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver_extension -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MZ -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/include -I../vendor/gateway-module-interface/lib -MMD -MF build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTConnectServer.o.d -o build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTConnectServer.o ../vendor/paho.mqtt.embedded-c/MQTTPacket/src/MQTTConnectServer.c -DXPRJ_TTN_Gateway_v1=TTN_Gateway_v1 -no-legacy-libc -DMQTT_TASK -G 0 
"/Applications/microchip/xc32/v1.42/bin/xc32-gcc" -g -D__DEBUG -D__MPLAB_DEBUGGER_PK3=1 -fframe-base-loclist -x c -c -mprocessor=32MZ2048EFM144 -ffunction-sections -O1 -D_FULL_SIGNAL_IMPLEMENTATION -I../src -I../vendor/harmony/framework -I../vendor/jsmn -I../vendor/json-builder -I../vendor/GenericRecorderLibSrc/Include -I../vendor/ttn-gateway-connector/src -I../vendor/protobuf-c -I../vendor/ttn-gateway-connector/src/github.com/TheThingsNetwork -I../vendor/ttn-gateway-connector/src/github.com/gogo/protobuf/protobuf -I../vendor/paho.mqtt.embedded-c/MQTTPacket/src -I../vendor/paho.mqtt.embedded-c/MQTTClient-C/src -I../vendor/mbedtls/include -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1 -I../src/TTN_Gateway_v1 -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1/framework -I../vendor/harmony/framework/driver/wifi/mrf24wn/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver_extension -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MZ -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/include -I../vendor/gateway-module-interface/lib -MMD -MF build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTDeserializePublish.o.d -o build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTDeserializePublish.o ../vendor/paho.mqtt.embedded-c/MQTTPacket/src/MQTTDeserializePublish.c -DXPRJ_TTN_Gateway_v1=TTN_Gateway_v1 -no-legacy-libc -DMQTT_TASK -G 0 
"/Applications/microchip/xc32/v1.42/bin/xc32-gcc" -g -D__DEBUG -D__MPLAB_DEBUGGER_PK3=1 -fframe-base-loclist -x c -c -mprocessor=32MZ2048EFM144 -ffunction-sections -O1 -D_FULL_SIGNAL_IMPLEMENTATION -I../src -I../vendor/harmony/framework -I../vendor/jsmn -I../vendor/json-builder -I../vendor/GenericRecorderLibSrc/Include -I../vendor/ttn-gateway-connector/src -I../vendor/protobuf-c -I../vendor/ttn-gateway-connector/src/github.com/TheThingsNetwork -I../vendor/ttn-gateway-connector/src/github.com/gogo/protobuf/protobuf -I../vendor/paho.mqtt.embedded-c/MQTTPacket/src -I../vendor/paho.mqtt.embedded-c/MQTTClient-C/src -I../vendor/mbedtls/include -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1 -I../src/TTN_Gateway_v1 -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1/framework -I../vendor/harmony/framework/driver/wifi/mrf24wn/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver_extension -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MZ -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/include -I../vendor/gateway-module-interface/lib -MMD -MF build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTSerializePublish.o.d -o build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTSerializePublish.o ../vendor/paho.mqtt.embedded-c/MQTTPacket/src/MQTTSerializePublish.c -DXPRJ_TTN_Gateway_v1=TTN_Gateway_v1 -no-legacy-libc -DMQTT_TASK -G 0 
"/Applications/microchip/xc32/v1.42/bin/xc32-gcc" -g -D__DEBUG -D__MPLAB_DEBUGGER_PK3=1 -fframe-base-loclist -x c -c -mprocessor=32MZ2048EFM144 -ffunction-sections -O1 -D_FULL_SIGNAL_IMPLEMENTATION -I../src -I../vendor/harmony/framework -I../vendor/jsmn -I../vendor/json-builder -I../vendor/GenericRecorderLibSrc/Include -I../vendor/ttn-gateway-connector/src -I../vendor/protobuf-c -I../vendor/ttn-gateway-connector/src/github.com/TheThingsNetwork -I../vendor/ttn-gateway-connector/src/github.com/gogo/protobuf/protobuf -I../vendor/paho.mqtt.embedded-c/MQTTPacket/src -I../vendor/paho.mqtt.embedded-c/MQTTClient-C/src -I../vendor/mbedtls/include -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1 -I../src/TTN_Gateway_v1 -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1/framework -I../vendor/harmony/framework/driver/wifi/mrf24wn/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver/include -I../vendor/harmony/framework/driver/wifi/mrf24wn/wireless_driver_extension -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MZ -I../vendor/harmony/third_party/rtos/FreeRTOS/Source/include -I../vendor/gateway-module-interface/lib -MMD -MF build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTConnectClient.o.d -o build/TTN_Gateway_v1/debug/_ext/2082537404/MQTTConnectClient.o ../vendor/paho.mqtt.embedded-c/MQTTPacket/src/MQTTConnectClient.c -DXPRJ_TTN_Gateway_v1=TTN_Gateway_v1 -no-legacy-libc -DMQTT_TASK -G 0 

...

"/Applications/microchip/xc32/v1.42/bin/xc32-gcc" -g -D__DEBUG -D__MPLAB_DEBUGGER_PK3=1 -fframe-base-loclist -x c -c -mprocessor=32MZ2048EFM144 -ffunction-sections -O1 -D_FULL_SIGNAL_IMPLEMENTATION -I../src -I../vendor/harmony/framework -I../vendor/jsmn -I../vendor/json-builder -I../vendor/GenericRecorderLibSrc/Include -I../vendor/ttn-gateway-connector/src -I../vendor/protobuf-c -I../vendor/ttn-gateway-connector/src/github.com/TheThingsNetwork -I../vendor/ttn-gateway-connector/src/github.com/gogo/protobuf/protobuf -I../vendor/paho.mqtt.embedded-c/MQTTPacket/src -I../vendor/paho.mqtt.embedded-c/MQTTClient-C/src -I../vendor/mbedtls/include -I../vendor/harmony/framework -I../src/system_config/TTN_Gateway_v1 -I../src/TTN_Gateway_v [... cut ...]
Firmware Suite Version.....01.45.41
Firmware type..............PIC32MZ

Target voltage detected
Target device PIC32MZ2048EFM144 found.
Device ID Revision = A3

Device Erased...

Programming...

The following memory area(s) will be programmed:
program memory: start address = 0x0, end address = 0xedfff
boot config memory
configuration memory
Programming/Verify complete
````

### Any Screenshots or Photos?

### Do you think you can try to fix this yourself and submit a pull request?

I would hope so.

KrishnaIyer, 2019-05-21:
@frankminfarm: This is almost certainly due to uploading a firmware without the bootloader. Once the compilation is done, please run `./merge_bootloader.sh` and flash the `firmware-with-bootloader.hex` using the PICKit.

The `firmware.hex` does not contain a bootloader and hence is intended only for using with an SD Card.

frankminfarm, 2019-05-21:
Yes, right you are. That was the problem. Thanks for the response.

KrishnaIyer, 2019-05-21:
👍

closed 2019-05-21 by KrishnaIyer