TheThingsProducts/gateway #60: Host header does not contain the port

adriansmares, 2019-04-04:
This is a bug report.

### What do you want to do? What do you expect?

While requesting the gateway configuration from the Account Server, the gateway does not specify the port in the `Host` header, even though it is a non-standard HTTP port.

### What steps did you take? (Give as many details as possible so that we can reproduce it)

Run an Account Server on port 1885.

### What went wrong? What is missing?

The gateway did not specify the port in the `Host` header while requesting the gateway configuration.

### What firmware is running on your gateway?

v1.0.5

### What do the logs show?

Not applicable.

### Any Screenshots or Photos?

Not applicable.

### Do you think you can try to fix this yourself and submit a pull request?

Not really - I think the HTTP stack doesn't fill the port correctly.

johanstokking, 2019-04-04:
Is this critical?

Can we work around this?

adriansmares, 2019-04-04:
If the v3 IdentityServer exposes the API on port 80 or 443, this issue can be avoided. Otherwise, servers running on different ports will generate the wrong frequency plan URL, due to the lack of port.

htdvisser, 2019-04-06:
Ref https://github.com/TheThingsNetwork/lorawan-stack/pull/436

PR #62 (2019-05-22): Add port to HTTP Host header
Closes #60
closed 2019-05-22 by htdvisser